Klipper3d/klipper #3376: Unknown command on Sidewinder X1

n3roGit, 2020-09-27:
I just installed Klipper using this tutorial and the linked config:

https://3dprintbeginner.com/install-klipper-on-sidewinder-x1/

Unfortunately I always get the following error message:

Unknown command:"M42"
Unknown command:"M42"
Unknown command:"M42"
Unknown command:"M117"

I guess I have to insert some code snippets to recognize the commands.

https://raw.githubusercontent.com/KevinOConnor/klipper/master/config/example-extras.cfg

Do you have a tip for me what I can do here and how?

If it works the complete config can also be included in your repo =)

Hywelmartin, 2020-09-27:
https://github.com/KevinOConnor/klipper/blob/55e20f2c198ae45485d714b4588fcf6768bc6165/config/example-extras.cfg#L1349
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

legendaryleo7, 2020-09-29:
@n3roGit, what slicer are you using? Can you attach your slicer start gcode, your printer.cfg and your logfile? 

Also, I have an artillery sidewinder x1 as well and found the tutorial you linked to not be super thorough. You will most likely need additional tweaking, although the issue you describe sounds like it's coming from your slicer. While klipper accepts many marlin commands, it is not marlin, and your slicer needs to be configured properly still, which was not adequately covered in the tutorial.

n3roGit, 2020-09-29:
Many thanks for the answer.

Currently I am using Cura 4.7 and Octoprint with the plugin OctoKlipper.

Which article should I ideally follow for the installation/setup?

GCodes:
[image]

Klipper cfg:
`[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^!ar3
position_endstop: 0
position_max: 310
homing_speed: 40.0

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^!ar14
position_endstop: 0
position_max: 310
homing_speed: 40
homing_positive_dir: false

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: ^!ar18
position_endstop: 0
position_min: -3
position_max: 400
homing_speed: 30
second_homing_speed: 1

[stepper_z1]
step_pin: ar36
dir_pin: !ar34
enable_pin: !ar30
step_distance: .0025

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: 0.002151191
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
min_extrude_temp: 200
min_temp: 5
max_temp: 300
max_extrude_only_distance: 500
max_extrude_cross_section: 2.0

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
min_temp: 5
[... 152 lines cut ...]
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
16:33:23 Shutdown due to M112 command
16:33:23 Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
16:33:23 Shutdown due to M112 command
16:33:23 Shutdown due to M112 command
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
16:33:23 Shutdown due to M112 command
16:33:28 Klipper state: Disconnect
16:33:31 Klipper state: Disconnect
Klipper state: Ready
16:33:36 Klipper state: Ready
16:34:04 Unknown command:"M42"
16:34:04 Unknown command:"M42"
16:34:04 Unknown command:"M42"
16:39:54 Unknown command:"M117"`

Hywelmartin, 2020-09-29:
M42 is For custom hardware not officially supported in Marlin, you can often just connect up an unused pin and use M42 to control it.
Klipper don't have that Gcode a similar function is [THIS](https://github.com/KevinOConnor/klipper/blob/55e20f2c198ae45485d714b4588fcf6768bc6165/config/example-extras.cfg#L1349)

M117 error is due to that You don't have any screen configured...

You also have gcode scripts in the slicer...

n3roGit, 2020-09-30:
Okay. I'm trying to figure this out.
currently i have entered the following into the config. but without success...

[output_pin my_pin]
pin: 7
value: 1.0

What to do to get out of 

M42 P4 S0 ; turn off the light
M42 P5 S0 ; turn off the light
M42 P6 S0 ; turn off the light

To make a klipper command?

n3roGit, 2020-09-30:
or like this is also wrong
[output_pin LED]
pin: 7
pwm: false
value: 0
shutdown_value: 0
#cycle_time: 0.001
#scale: 1000

[gcode_macro M42]
default_parameter_S: 1000
default_parameter_P: 100
gcode:
    SET_PIN PIN=LED VALUE=200

legendaryleo7, 2020-10-01:
> or like this is also wrong
> [output_pin LED]
> pin: 7
> pwm: false
> value: 0
> shutdown_value: 0
> #cycle_time: 0.001
> #scale: 1000
> 
> [gcode_macro M42]
> default_parameter_S: 1000
> default_parameter_P: 100
> gcode:
> SET_PIN PIN=LED VALUE=200

1) I think it's `ar7`, since your config file is using arduino style mapping.
2) `VALUE` should be 0 or 1. If you have `pwm` enabled, you can use a floating number between 0 and 1 as well. The 0-255 scale is not supported.

n3roGit, 2020-10-02:
Now i have a working config:
`[output_pin LED_Green]
pin: ar4
pwm: false
value: 0
shutdown_value: 0

[output_pin LED_Red]
pin: ar5
pwm: false
value: 0
shutdown_value: 0

[output_pin LED_Blue]
pin: ar6
pwm: false
value: 0
shutdown_value: 0

[gcode_macro M42]
gcode:
    SET_PIN PIN=LED_Green VALUE=1.0
    SET_PIN PIN=LED_Blue VALUE=1.0
    SET_PIN PIN=LED_Red VALUE=1.0
[gcode_macro M43]
gcode:
    SET_PIN PIN=LED_Green VALUE=0.0
    SET_PIN PIN=LED_Blue VALUE=0.0
    SET_PIN PIN=LED_Red VALUE=0.0`

It is not as beautiful as I would like it to be but it does what it should.
I would like to have M42 for a scaling between 0-255

Hywelmartin, 2020-10-02:
[output_pin LED_Green]
pin: ar4
pwm: true
value: 0
shutdown_value: 0

[gcode_macro M42]
default_parameter_S: 255
gcode:
            SET_PIN PIN=LED_Green VALUE={ {S} / 255 }

format like M42 S=180

might work

n3roGit, 2020-10-02:
Thank you I will try this.

It's only M42 because of the stock firmware name.
You naming in better thanks

n3roGit, 2020-10-02:
This works but the led is blinking when its not 255. no dimming only different blink times

Hywelmartin, 2020-10-04:
Alter the PWM frequency

closed 2020-10-04 by klipper-gitissuebot