Klipper3d/klipper #3378: Command format mismatch

Oranginator, 2020-09-27:
Hi guys,

updated klipper to the latest version, now I'm getting the following error: 

Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u
This type of error is frequently caused by running an older version of the firmware on the micro-controller (fix by recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

I updated the klipper firmware on SD to the latest version.

Thanks in advance for helping me.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5289074/klippy.log)

Arksine, 2020-09-27:
The log shows that you need to build and flash the latest version of klipper to the MCU.  You mention that you attempted this, however it must not have flashed correctly.

Oranginator, 2020-09-27:
I tried serveral times, as it's an SKR 1.1 Pro i only need to copy the klipper.bin to the SD-Card and rename it to FIRMWARE.CUR, as I did with all other versions before without having issues. I now tried it with renaming it to firmware.bin.
Make clean also didn't solve the problem.

closed 2020-09-27 by Oranginator