Klipper3d/klipper #3384: Neopixel control is not working over RPi Microcontroller

altvnk, 2020-09-29:
I have a WS2811 LED strip with total of 48 LED's connected to RPi output pin GPIO16. My `printer.cfg` configuration for Neopixel:
```
[neopixel case]
pin: rpi:gpio16
chain_count: 48
initial_RED: 1
initial_GREEN: 1
initial_BLUE: 1
```
During Klipper startup I can see the message `"Neopixel update did not succeed"`. And only one, very first LED lights up green. Sometimes, when Linux MCU initialises, LED strip lights up white, no matter how initial colours are configured, but when I reconnect the strip and issue SET_LED command the same LED becomes green and `"Neopixel update did not succeed"` shows up in the log. Currently I'm on branch which have support for 85 total LEDs in row, but behaviour is not changed compared to the master branch.

I don't believe it's a wiring issue, since this setup works perfectly fine when controlled with Python code directly using Adafruit neopixel library.

Attaching log file, so  bot can be happy.
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/5301931/klippy.log.gz)

KevinOConnor, 2020-09-30:
See the recent discussion in #3366 .

-Kevin

altvnk, 2020-09-30:
Thanks, have no idea how I missed that ;)

closed 2020-09-30 by altvnk