Klipper3d/klipper #3385: Timeout during homing

PureTek-Innovations, 2020-09-30:
My printers firmware keeps reporting an error:

Failed to home probe. Timeout during homing.

I have dual z stepper motors and a BL touch.

When I run, G28 then Z_TILT_ADJUST then BED_MESH_CALIBRATE then I get this error. It's driving me nuts. Sometimes it won't even manage to complete Z_TILT_ADJUST.

I've increased all of the timeouts by at least a factor of 10 and set the maximum consecutive timeouts to 0 (disabled).

Any and all help appreciated. Thanks

[klippy logs.zip](https://github.com/KevinOConnor/klipper/files/5305998/klippy.logs.zip)

ld3300, 2020-09-30:
Hi @PureTek-Innovations,

Whatever is going on in the serial.log file is weird.  I'm not sure how it affects things in reality, but why are there G91 commands happening before your G28 commands?

-David

PureTek-Innovations, 2020-09-30:
I have no idea why the G91 commands are there, they appear to have been added by Klipper, I just type G28 in the terminal...

ld3300, 2020-09-30:
ahh, Klipper doing it might make sense.

It looks like you have put a z-tilt section in your config, but because you don't have an endstop defined for Z1 it is never able to home.

PureTek-Innovations, 2020-09-30:
I seem to remember that adding "endstop_pin = probe:z_virtual_endstop" under [stepper_z1] threw an error, so I took it out. I can try putting it back in.

ld3300, 2020-09-30:
I'm not sure z-tilt is what you actually want.  Z-tilt would be for 2 z motors that each have their own endstop so the machine can level them out.  Is it bed_tilt you are looking for instead?

PureTek-Innovations, 2020-09-30:
I'm not sure, but Z_TILT_ADJUST normally works, it usually fails on the bed mesh calibrate

PureTek-Innovations, 2020-09-30:
I added in "endstop_pin = probe:z_virtual_endstop" under [stepper_z1] and I get this error:
 pin z_virtual_endstop used multiple times in config
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready

I'll take it out again

PureTek-Innovations, 2020-09-30:
Found the problem, it's a misleading error message. The Z_Probe was going out of range, I recently rebuilt the hot end and moved the probe. Decreased both position_min: -5 to -10 and it all works again :)

PureTek-Innovations, 2020-09-30:
Dear Developers is it possible for this error message to be fixed? Maybe something like an Out Of Range error?

FHeilmann, 2020-10-01:
While I agree that it's possibel to send users the wrong way by calling it timeout, it is an accurate error message.
Basically the klipper host code tells the printer to start homing and based on the range of motion on that axis, estimates how long this homing move will take. If during this time, the host code doesn't get any report from the mcu that the homing move is finished, it throws the error. So `timeout` is still an accurate description here.

The same error can be thrown when, for example, the microsteps are misconfigured or the axis is not moving for some reason. Calling the error `Out of Range` would be inaccurate for this case.

PureTek-Innovations, 2020-10-02:
Okay, thank you for the clarification.

closed 2020-10-02 by PureTek-Innovations