Klipper3d/klipper #3387: RPi 3 multiple mcu randomly throws "MCU 'host' shutdown: Stepper too far in past"

tntclaus, 2020-10-01:
Hi, I have multiple MCU corexy assembly where RAMPS is controlling z-steppers and extruder (mcu `mcu`) and RPi3 controlling x/y steppers (mcu 'host') and x/y steppers use x16 micro stepping. 

That is RPi3 mcu that fails, not RAMPS/Arduino. 

When I slice gcode in e.g. Cura (or Slic3r) using speeds above 60 mm/s I start getting random "Stepper in past" errors. It may happen in the middle of the print, in the beginning, or in the end. 

Attached is the `klippy.log` of such a print (and `printer.cfg), anywhere in gcode the speed does not exceed `F12000` (200 mm/s). This print fails in the first 1/3 of the model somewhere after the speed was set to `F9000`. 

I wonder why it was able to go `F12000` in this gcode many times before it finally failed? And why that happens in random places of the same gcode?

My setup of RPi3 can definitely keep speeds up to `F36000` (tested with custom gcode that moves toolhead along perimeters and diagonals, btw this gcode is also sometimes randomly fails, sometimes it doesn't). 

Right now the problem seems for me like a software bug that occurs on RPi. If it is I can try to fix it myself, but I will need help bcz the codebase is somewhat big an unfamiliar to me. 

[printer.cfg.1.txt](https://github.com/KevinOConnor/klipper/files/5311232/printer.cfg.1.txt)
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5311163/klippy.log.zip)

KevinOConnor, 2020-10-02:
The "linux mcu" code is not intended to be a high-performance system.  (It uses a slow linux kernel interface to toggle gpio pins.)  In addition to not being able to reach very high speeds, the individual step pulses are likely to not have precise timing either.

The "linux mcu" code is more intended for minor background type tasks (turning on leds, fans, power relays, etc.).  

-Kevin

tntclaus, 2020-10-07:
@KevinOConnor that's what I thought firs, but in my case it doesn't seem like `klipper_mcu` process has no spare cpu cycles. I already made an experimental patch for RPi3 which utilises direct memory access to handle gpio control (x70 times performance gain from somewhat 0.6 to 36 MHz). 

This actually didn't help a lot as there is another inefficient system call of `get_system_clock` function (or something like that), in RPi3 it uses ARM timer (is probably not precise enough and can be throttled) instead of BCM timer (which has us resolution and can't be throttled). Unfortunately it was not as easy as controlling gpio, as after I switched to this timer with direct memory access the communication between `klippy` and `klipper_mcu` became impossible. 

If you could give me any hints why this may happen I probably would be able to solve this myself and introduce a pull request with mcu firmware for RPi3/4 devices. 

Right now I tried a dirty hack: just commented out "stepper too far in past" shutdown and, well, I don't see any failures in models that I print. Prints are absolutely fine and I can run as high as 300 mm/s without any issues. But this is really not good, as the "stepper too far in past" has its purpose.

KevinOConnor, 2020-10-07:
> I already made an experimental patch for RPi3 which utilises direct memory access to handle gpio control (x70 times performance gain from somewhat 0.6 to 36 MHz).

Interesting.

> This actually didn't help a lot as there is another inefficient system call of get_system_clock function (or something like that),

The `clock_gettime(CLOCK_MONOTONIC, &ts);` call shouldn't be that bad as it uses the vDSO on the RPi.  It's not super fast (there's some multiplication / division on each call to calculate the time), but I'd be surprised if it caused issues.

> Unfortunately it was not as easy as controlling gpio, as after I switched to this timer with direct memory access the communication between klippy and klipper_mcu became impossible.

It should be possible to use a direct register read for the timer, but then you'd have to make sure it does the correct thing during linux cpu frequency changes.  It would also be necessary to calculate the appropriate sleep time in `console_sleep()`. Not sure if you've already done that or not.

> Right now I tried a dirty hack: just commented out "stepper too far in past" shutdown and, well, I don't see any failures in models that I print. Prints are absolutely fine and I can run as high as [...]

FWIW, I know of the following timing issues on the Linux MCU code:
1. The code doesn't have "strict timing" because Linux doesn't have "strict timing".  Some things to check:
  * Make sure you're running the linux_mcu code with the `-r` option so that it uses real-time scheduling.
  * You could always increase the timeout in the "stepper too far in the past" error from 1ms to 5ms (eg, `timer_from_us(5000)`).  It's possible that basic Linux is struggling to make a 1ms time window (though I'd find that odd for code running with real-time scheduling).  You could also add debugging here (eg, `if ((int32_t)(s->next_step_time - min_next_time) < (int32_t)(-timer_from_us(1000))) output("time close %u %u", s->next_step_time, min_next_time);`) and see how far off the timers are.  Having the schedule off by a few ms probably wont cause any notable harm, but much more than that and I'd say things wont work as they should.
  * It's possible to try with a real-time Linux kernel.  I don't know if that would improve things or not.
  * The linux mcu code probably should lock it's ram and code into memory so that it is never swapped out.  I'd say this is unlikely to be the cause of your problem though.  Even with ram locking there would still be variable timing due to page cache misses and cache line misses.
2. The code uses CLOCK_MONOTONIC when it really should use CLOCK_MONOTONIC_RAW.  Unfortunately, a call with CLOCK_MONOTONIC_RAW doesn't use the vDSO, and thus would actually increase processing load (it would require a Linux syscall on each time read).  I don't think this would cause the errors you are seeing.  As a test though, you could try disabling ntp so that CLOCK_MONOTONIC effectively works like CLOCK_MONOTONIC_RAW.
3. The code polls for timers instead of using "interrupts".  It's possible to convert the code to use Unix signals to emulate interrupts.  I actually coded this up a couple of months ago.  However, using signals results in a system call on every `irq_enable()` / `irq_disable()` sequence and these system calls don't use the vDSO, so I fear they would reduce overall performance.  I can share this code with you if you want to test it though.  I think it should be possible to use signals without the `irq_disable()` / `enable_irq()` syscall overhead, but I have not attempted to code that.

-Kevin

tntclaus, 2020-10-08:
Here is commit with modified `gpio.c` that currently runs on my printer. The code is not clean (and I am not a C/C++ developer though ;) ), as it supposed to be just an experiment:

https://github.com/tntclaus/klipper/commit/ca07a4bc8782319b14ad6616459e1ce5b4295b0e

> The clock_gettime(CLOCK_MONOTONIC, &ts); call shouldn't be that bad as it uses the vDSO on the RPi. It's not super fast (there's some multiplication / division on each call to calculate the time), but I'd be surprised if it caused issues.

I found [an investigation](https://yosh.ke.mu/article/raspberry_pi_getting_time) that direct call to memory register is about 8 times faster than system call. I didn't benchmark it myself but trust it as it complies with benchmarks I did myself with GPIO direct access. If `CLOCK_MONOTONIC` used vDSO in 2015 than the benchmark at cited article should be relevant for modern day kernel too.

> It should be possible to use a direct register read for the timer, but then you'd have to make sure it does the correct thing during linux cpu frequency changes. It would also be necessary to calculate the appropriate sleep time in console_sleep(). Not sure if you've already done that or not.

Thanks for this hint. I guess I didn't touch `console_sleep()` function, so I will check it. 

> Make sure you're running the linux_mcu code with the -r option so that it uses real-time scheduling.

That's done. It was the first thing I checked in sources when I faced the problem. 

> You could always increase the timeout in the "stepper too far in the past" error from 1ms to 5ms (eg, timer_from_us(5000)).

Will try that.

> The code uses CLOCK_MONOTONIC when it really should use CLOCK_MONOTONIC_RAW. Unfortunately, a call with CLOCK_MONOTONIC_RAW doesn't use the vDSO

Interesting. I think I will do benchmark these things first (I mean direct timer access, and various clock calls). If there will be really no difference with clock calls then something else must be there that uses system space too much and probably can be optimised.

Here is the code of direct timer call for RPi3:

```
#include <stdio.h>
#include <stdlib.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <unistd.h>

#define BCM2708_ST_BASE 0x3F003000 /* BCM 2835 System Timer */

volatile unsigned *bcm2835_timer_registers = NULL;

/*
 * @see https://yosh.ke.mu/article/raspberry_pi_getting_time
 */
void
bcm2835_timer_init(void)
{
    if(bcm2835_timer_registers)
        return;

    /* open /dev/mem */
    int TIMER_memFd;
    if ((TIMER_memFd = open("/dev/mem", O_RDWR/*|O_SYNC*/) ) < 0)
    {
        shutdown("can't open /dev/mem - need root ?\n");
    }

    fprintf(stderr, "bcm2835_timer_init\n");

    /* mmap BCM System Timer */
    void *TIMER_map = mmap(
            NULL,
            4096, /* BLOCK_SIZE */
            PROT_READ /*|PROT_WRITE*/,
            MAP_SHARED,
            TIMER_memFd,
            BCM2708_ST_BASE
    );

    close(TIMER_memFd);

    if (TIMER_map == MAP_FAILED)
    {
        printf("mmap error %d\n", (int)TIMER_map);
        exit(-1);
    }
    bcm2835_timer_registers = (volatile unsigned *)TIMER_map;
}

unsigned int
bcm2835_timer_ticks(void) {
    return bcm2835_timer_registers[1];
}
```

> However, using signals results in a system call on every irq_enable() / irq_disable() sequence and these system calls don't use the vDSO, so I fear they would reduce overall performance

I guess it may reduce performance. Let me check if timer optimisation is possible and what we can gain if it is ;)

KevinOConnor, 2020-10-08:
Interesting.

FYI, I uploaded my test linux mcu branch that uses signals instead of polling: https://github.com/KevinOConnor/klipper/tree/work-linux-irq-20200607

`cd ~/klipper ; git fetch ; git checkout origin/work-linux-irq-20200607 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start`

-Kevin

tntclaus, 2021-04-10:
Hi, @KevinOConnor, didn't have time to work on this but it seems like now I've finally got some time. Can you please share once again your irq branch? The link doesn't work now.

KevinOConnor, 2021-04-10:
Sure - I pushed that branch back up.

-Kevin

tntclaus, 2021-04-10:
Great, thanks!

tntclaus, 2021-04-13:
@KevinOConnor I plan to introduce pull request with native GPIO mode for RPi  3/4 and therefore there is a question how to do it. I see two options:

1. Introduction of additional controller type in menuconfig besides Linux and name it RPi 3/4. This will use 
2. Autodetect RPi hardware and silently enable native gpio mode. 

Your opinion is very important for me. Personally I tend for option 1.

tntclaus, 2021-04-13:
Actually I chose 3rd way: added a selector with low level options for either generic or native rpi2/3 gpio access. It defaults to generic:
https://github.com/tntclaus/klipper/commit/125ea44d4b6076c33d8f6e559a25dd773e224727

Besides that I introduced low level option to disable stepper too far in past check with appropriate warning:
https://github.com/tntclaus/klipper/commit/3b89c196c4933f8f2ecf07ff63d58c7ea3c2b57e

The latter is useful at least for me as I am hacking with various motors, drivers and controllers; so if you don't mind this will be great to keep it as an option too :)

tntclaus, 2021-04-13:
I tried you irq  branch with my native gpio implementation and it works like a charm: 

* sys load dropped from somewhat 80% average using timer and to 40% average;
* `klipper_mcu` process consumes 50% cpu in average vs 95% cpu without irq.

I consider this to be awesome result.

Right now I am doing test print on speed 150mm/s with x/y motors controlled by raspberry and everything is fine. Next time I will enable shutdown for "stepper too far in past" and see what happens.

tntclaus, 2021-04-13:
Well, I guess this opens interesting possibilities for designing ramps-like shields for raspberries that can handle all the kinematics alone.

This may require running multiple klipper_MCU instances at same RPI host.

KevinOConnor, 2021-04-14:
Great!

> Actually I chose 3rd way: added a selector with low level options for either generic or native rpi2/3 gpio access. It defaults to generic

That's fine with me.

> Besides that I introduced low level option to disable stepper too far in past check with appropriate warning:

That's okay for local testing, but would not be a good candidate for the master klipper branch.  Disabling that check would prevent the benchmarks from running, and it could lead to cases where the schedule significantly deviates without any notification.

> klipper_mcu process consumes 50% cpu in average vs 95% cpu without irq.

Interesting.

-Kevin

tntclaus, 2021-04-14:
Hi, i did some benchmarks for 2 stepper configuration:

```
allocate_oids count=2
config_stepper oid=0 step_pin=gpio27 dir_pin=gpio18 min_stop_interval=0 invert_step=0
config_stepper oid=1 step_pin=gpio2 dir_pin=gpio17 min_stop_interval=0 invert_step=0
finalize_config crc=0
```

```
SET start_clock {clock+freq}
SET ticks 950

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=Eval: SET start_clock 13889478616
0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0Eval: reset_step_clock oid=0 clock=13889478616
Eval: queue_step oid=0 interval=950 count=60000 add=0
Eval: reset_step_clock oid=1 clock=13889478616
Eval: queue_step oid=1 interval=950 count=60000 add=0

163.934: stats count=61 sum=40159526 sumsq=4294967295
169.004: stats count=67 sum=44597 sumsq=250921
174.104: stats count=57 sum=45278 sumsq=367782
ECHO Test result is: {"%.0fK" % (3. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 158K

SET start_clock {clock+freq}
SET ticks 700

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=Eval: SET start_clock 327506987490

Eval: reset_step_clock oid=0 clock=32750698749
Eval: queue_step oid=0 interval=700 count=60000 add=0

543.105: stats count=61 sum=223235 sumsq=34640421
ECHO Test result is: {"%.0fK" % (3. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 214K
```

Single stepper 214K (probably can be more), and double stepper 158K (i think can be more).

I changed timing toleration from 1000 to 5000 in both stepper.c and timer_irq.c to get these results.

tntclaus, 2021-04-14:
Interesting thing is that running same benchmark multiple times in a row may lead to "stepper too far in past" or "rescheduled" messages. So I spent some more time bisecting ticks, here is the list of rather safe values (10 successful times in a row) that I've got on my RPi 3:

* 1 stepper, ticks = 670, 224K
* 2 steppers, ticks = 980, 153K

I guess result for 3 steppers will be somewhere around 120K. I can't test it right now. 

I wonder if using RPi RT kernel can boost performance even more? Stepper in past events are brought mostly from random delays because of mcu process being interrupted by scheduler in favour of some other task. 

On the other hand — do I (or anyone else running hobby machine) really needs any better performance? What do you think?

KevinOConnor, 2021-04-14:
FYI, the `3 *` in the `ECHO` command needs to be updated when using 1 or 2 steppers.  So, 1 stepper at 670 ticks is actually `50000000 / 670 ~= 75K`.  Two steppers at 980 ticks is ~102K.

When I last ran the benchmarks (see docs/Benchmarks.md) I was getting twice the update rates you're reporting now.  So, something doesn't look right.

> I had to change timing toleration from 1000 to 5000 in both stepper.c and timer_irq.c to get these results.

I'm okay with that change for the "Linux" builds.  (We can check for CONFIG_HAVE_STRICT_TIMING and increase to 5000us in that case.)

> I wonder if using RPi RT kernel can boost performance even more? Stepper in past events are brought mostly from random delays because of mcu process being interrupted by scheduler in favour of some other task.

If the process is started with `-r` then Linux shouldn't schedule some other task.  That said, it is possible hardware irqs, kernel events, cache misses, etc. could cause scheduling delays.  An RT kernel would likely improve that, but it's not something I've tested or looked at.

-Kevin

tntclaus, 2021-04-14:
> When I last ran the benchmarks (see docs/Benchmarks.md) I was getting twice the update rates you're reporting now. 

Interesting. I will do same tests with current upstream. Do you have anywhere console commands you used to do the benchmark? And which device revision did you use?

I also just started 4 hour (150mm/s, 0.005mm/step, 5k accel) real print with stepper too far in past shutdown enabled. If it will finish successfully then it will be indication of progress (as it always failed above 60 mm/s speeds on upstream branch).

KevinOConnor, 2021-04-14:
> Do you have anywhere console commands you used to do the benchmark?

https://www.klipper3d.org/Benchmarks.html#linux-mcu-step-rate-benchmark

-Kevin

tntclaus, 2021-04-16:
Here is the graph of RPi host load for short 150 mm/s print, I don't understand what it says really and it does not match top/htop data for mcu core load (which was somewhat 2 times higher). 

[image]

Btw, 4 hour print with **enabled** "stepper too far in past shutdown" with 5k accels and 150mm/s speed finished perfectly: part was printed exactly how it was sliced. As I said before this patch i couldn't print with speeds above 60mm/s.

**So what we have now:**

1. `klipper_mcu` CPU load reduced twice from ~95% to ~50% for 150 mm/s speeds, and simple approximation leads me to conclusion that ~100% load will be around 300 mm/s speeds;
2. "stepper too far in past" error during real prints does not fire anymore;
3. benchmarks degraded 3 times from what it had been at commit `06437c58`;
4. corrected benchmark for current commit 2 stepper configuration is `158K/3*2 = 105K`, which should provide speeds up to `0.005 * 105000 = 525 mm/s` for double stepper moves (straight lines) — considering real life mcu ONLY controls X/Y steppers, this doesn't match point 1.

**Questions:**

1. How is it possible to have 3 times better benchmarks for at least 2 times worse performance? Poor understanding from my side (yes, I really not quite understand how it works internally)? Bugs in benchmarking for linux icu? 
2. Should we spend time determining the reason why this happens or is it enough? 

While I can't answer first question, the second one is quite simple: practical result seems good for me. I'd prefer right now not to find out why benchmarks doesn't match real life :) If you agree I would proceed in following way in preparing the pull request:

1. For `MARCH_LINUX` new low level options:
2. 1. add selection `Timer implementation` with options: `irq`, `polling`, defaults to `polling` (current behaviour);
2. 2. introduce selector `GPIO Access` for either generic or native rpi2/3, defaults to `generic`
2. 3. introduce int setting for changing `timer_from_us(1000)` from default `1000` to any other value, which defaults to `1000` for personal tuning.

KevinOConnor, 2021-04-16:
> I don't understand what it says really and it does not match top/htop data for mcu core load (which was somewhat 2 times higher).

The micro-controller load reporting uses a heuristic based on observed jitter of mcu task scheduling.  I would not expect it to match the Linux process accounting - they measure different things.

> How is it possible to have 3 times better benchmarks for at least 2 times worse performance?

My guess is that there is a bug in the new scheduling code, or that we're running the benchmarks in a different way.  Were you able to reproduce the previous benchmark results (when using the master branch)?

> Should we spend time determining the reason why this happens or is it enough?

I would not feel comfortable making a change without understanding why the benchmarks have changed.

That said, if we can track down the reason and the new code still produces better real world results with lower benchmarks then I'd be inclined to commit it.  That is, the benchmarks are a tool for finding problems and measuring progress; they aren't a goal in and of themselves.

> add selection Timer implementation with options: irq, polling, defaults to polling (current behaviour);

No - that isn't a choice a user can reasonably make.  We'll either remain with polling or go forward with signals.

> introduce selector GPIO Access for either generic or native rpi2/3, defaults to generic

That's fine, but I would word it as a "Processor Model" selection so that it fits the layout of other architectures in "make menuconfig".

> introduce int setting for changing timer_from_us(1000) from default 1000 to any other value, which defaults to 1000 for personal tuning.

No - this isn't a choice a user can reasonably make.  If we can use the same 1ms for all architectures that's great.  If we have to increase it to 5ms for the Linux architecture then that's okay.  If it needs to be something else, then I think we need to track down why.

Cheers,
-Kevin

tntclaus, 2021-04-16:
> Were you able to reproduce the previous benchmark results (when using the master branch)?

Yes, they are the same on current master branch and what you expected to have. See log below.

Version benchmarked: `Loaded 98 commands (v0.9.1-425-gca1d4316-20210416_203600-fluiddpi / gcc: (Raspbian 8.3.0-6+rpi1) 8.3.0 binutils: (GNU Binutils for Raspbian) 2.31.1)`

```
allocate_oids count=2
config_stepper oid=0 step_pin=gpio27 dir_pin=gpio18 min_stop_interval=0 invert_step=0
config_stepper oid=1 step_pin=gpio2 dir_pin=gpio17 min_stop_interval=0 invert_step=0
finalize_config crc=0
038.696: stats count=59 sum=71802 sumsq=2087240
SET start_clock {clock+freq}
SET ticks 356

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0Eval: SET start_clock 3828357476
Eval: reset_step_clock oid=0 clock=3828357476
Eval: queue_step oid=0 interval=356 count=60000 add=0
Eval: reset_step_clock oid=1 clock=3828357476
Eval: queue_step oid=1 interval=356 count=60000 add=0
043.796: stats count=65 sum=57343 sumsq=431460

ECHO Test result is: {"%.0fK" % (2. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 281K
```

The problem with this benchmark is that it is in no way resembles how it really works under RPi: absolutely impossible to have 281K (way above 60 mm/s) steps for two steppers per second. Real life maximum is 12K (60 mm/s move along X or Y) until it dies with `stepper to far in past`. ~20 times deviation is much worse than ~2 times deviation vs real life performance that we saw with irq branch benchmarks. 

I have no idea why this happens. Probably bug was introduced. Probably more than one :) Probably some bugs were already there and Linux arch was not benchmarked correctly from the very beginning.  Probably irq branch even fixed something :)

> if we can track down the reason

Let's try, but I can't do it all alone :) If you direct me where to look — I will spend some time this weekend tracking down the problem.

> No - that isn't a choice a user can reasonably make. We'll either remain with polling or go forward with signals.

Regarding irq/polling selector: I agree that it is not a choice a user can reasonably make. On the other hand my experience with multiplatform software development taught me that sometimes old solution is more reliable than new one and it is not always obvious when new one becomes stable enough. Giving user an opportunity to chose between new way and old way is always better. If you ask me I would go first introducing irq as experimental feature, with polling enabled by default, then — making it a default feature leaving polling as fallback option for those who may face problems with irq. And some time later — discontinuing polling if it proves its stability and efficiency over time. 

> No - this isn't a choice a user can reasonably make. If we can use the same 1ms for all architectures that's great. 

For me from the practical point even 1 second looks good: remember, I had running printer for days with "stepper too far in past" shutdown disabled and had no problems ever. But for someone else this timing may be very critical, one may want to set it as strict as possible for whatever reason I even can't imagine. 

IMHO options like these, they are not for users, they are for hackers (like me). One may still need some easy way to tweak firmware behaviour without changing sources. That becomes a bit pain in the ass when you patched code and need to pull an update, setup another printer, etc. Probably we can introduce experimental / for developers only section of menuconfig where such settings could be? Most of users don't try to mess with unknown options, especially when they are marked with DANGER! DO IT ON YOUR OWN RISK!. That's my opinion in general.

KevinOConnor, 2021-04-17:
I took a look at the code, and I was doing something silly on the work-linux-irq-20200604 branch.  (The irq_disable()/irq_enable() code is expensive on that branch, and the new code was unnecessarily calling it on every software timer dispatch.)  I've made the fix and rebased that branch.  The benchmark results on that branch are now similar to the benchmarks on the master branch in my tests.

> If you ask me I would go first introducing irq as experimental feature, with polling enabled by default, then — making it a default feature leaving polling as fallback option for those who may face [...]

I understand, but I prefer to keep experimental code on git branches (and/or separate git trees) until we feel it is no longer experimental.

> IMHO options like these, they are not for users, they are for hackers (like me)

I understand, but again I prefer to use git branches for experimental work.

Cheers,
-Kevin

tntclaus, 2021-04-17:
> I understand

Got it :) 

I did some profiling of console benchmarking today. Every benchmark was stopped when first shutdown was hit. The results are NOT consistent as from profile to profile you may get different call time table, but it may be of some interest. 

These are irq and master branches:
[klipper_mcu_profile_irq_generic_gpio_200ticks_558cceaa.txt](https://github.com/KevinOConnor/klipper/files/6329798/klipper_mcu_profile_irq_generic_gpio_200ticks_558cceaa.txt)
[klipper_mcu_profile_polling_generic_gpio_350ticks_c5f1835b.txt](https://github.com/KevinOConnor/klipper/files/6329801/klipper_mcu_profile_polling_generic_gpio_350ticks_c5f1835b.txt)

According to profiler analysis for current master `irq_poll` is the most expensive call, which comply with my observations. 

For your irq branch `irq_disable` and `irq_enable` calls are not expensive at all (but why? especially if the fix benchmarks?) and `gpio_out_write` calls come to first place at consuming cpu cycles (that makes sense, as ioctl calls on rpi are known to be inefficient).  

Comparing two profiling results it seems sane to make an assumption that irq is more efficient than polling in terms of effectively consuming CPU cycles. So it should be preferred. 

These are profiles from my working branch using irq, in generic and native gpio:
[klipper_mcu_profile_irq_generic_gpio_200ticks_dirty.txt](https://github.com/KevinOConnor/klipper/files/6329799/klipper_mcu_profile_irq_generic_gpio_200ticks_dirty.txt)
[klipper_mcu_profile_irq_native_gpio_200ticks_dirty.txt](https://github.com/KevinOConnor/klipper/files/6329800/klipper_mcu_profile_irq_native_gpio_200ticks_dirty.txt)

What is weird: in my current branch gpio calls look considerably cheaper (comparing to current master), both using `ioctl` and native calls. 

`timer_dispatch` starts to consume a lot of time and it is consistent between profiling rounds. Moreover: all in all it seems less efficient than current master. Consider this (cumulative for whole run):

* `239999` calls of `gpio_out_toggle_noirq` in current master branch took only `0.05` seconds, almost same result with your previous revision of irq branch (`0.06` scones for whole run).
* `239593` calls of `gpio_out_toggle_noirq` in my dirty branch took `0.09` seconds when using `native` gpio calls and `irq`. This is almost twice worse than current master.
* `197246` calls of `gpio_out_toggle_noirq` in my dirty branch took `0.12` seconds when using `ioctl` gpio calls. This is about 3 times worse than current master (and it matches our observations with). 

I wonder if I merged something wrong?

Anyway seems like if we want to do maximum optimisation we will have to try direct memory clock calls as `timer_dispatch` does clock calls down on stack. 

Tomorrow I will continue to do profiling. Will check your updated irq branch and merge it with my gpio.

KevinOConnor, 2021-04-17:
It looks like you tried to use gprof to profile the code?  Alas, that tool is not accurate enough for high-performance code like this.  What ends up happening is the cost of profiling itself skews the numbers so dramatically that it results in misleading information.  It also doesn't accurately account for system calls.

A much better tool is `perf`.  (It can still skew the results, but in general it does a much better job.)  It's typically installed with something like `sudo apt-get update ; sudo apt-get install linux-perf`.  Then profiling is done with something like `sudo perf record ./out/klipper.elf` and (after exiting the program) the results viewed with `sudo perf report`.  There's a bunch of options available and one can find further info online.  Alas, the raspbian distro can be quite confusing with installation of perf - sometimes it's necessary to do something like `sudo apt-get update; sudo apt-get install linux-perf-4.18` and then capture data with `sudo perf-4.18 record ./out/klipper.elf`.

> p.s. what is also may be of some interest: trying to fire benchmark with 200 ticks leads to Move queue overflow on current master

I can't reproduce that behaviour.  It would generally indicate a memory corruption bug - or maybe a bad compile?

-Kevin

tntclaus, 2021-04-18:
Thanks, I'll try perf. This is my first profiling in Linux :) 

> maybe a bad compile?

I guess it's. Gprof requires binary to be compiled with `-gt` option.

tntclaus, 2021-04-18:
I did following benchmarks today on native pgio + polling: 

Best results are:
* 2 stepper, 237 ticks, 422K
* 1 stepper, 230 ticks, 217K

Console commands for 2 stepper benchmark:
```
SET start_clock {clock+freq}
SET ticks 237

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0

Eval: SET start_clock 33976713110
Eval: reset_step_clock oid=0 clock=33976713110
Eval: queue_step oid=0 interval=237 count=60000 add=0
Eval: reset_step_clock oid=1 clock=33976713110
Eval: queue_step oid=1 interval=237 count=60000 add=0

321.344: stats count=65 sum=53255 sumsq=463193
326.444: stats count=56 sum=30897 sumsq=226183
ECHO Test result is: {"%.0fK" % (2. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 422K
```

Console commands for 1 stepper benchmark:
```
SET start_clock {clock+freq}
SET ticks 230

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=Eval: SET start_clock 452535916910

Eval: reset_step_clock oid=0 clock=45253591691
Eval: queue_step oid=0 interval=230 count=60000 add=0
548.345: stats count=61 sum=43764 sumsq=380542
ECHO Test result is: {"%.0fK" % (1. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 217K
```

tntclaus, 2021-04-18:
Benchmarking of `irq + native gpio` has slightly worse results than `polling + gpio`, but still better than `irq` + `ioctl`. From my point of view they are of no interest. 

What is really interesting: CPU consumption. Here is the list of ticks — CPU core load (average of multiple runs):

**For polling**
| Ticks  | CPU Load, % |
| -----------: | ------------: |
|  300  |  ~28 |
|  500  |   ~50|
|  800  |   ~75 |
| 1000  |   ~90 |
| 1200  |   ~90 |
| 1400  |   ~90 |
| 1600  |   ~80 |
| 1800  |   ~75 |
| 2000  |   ~70 |

**For IRQ**
| Ticks  | CPU Load, % | Comment | 
| -----------: | ------------: | ------------ |
|  300  |  ~30 | May throw `Stepper too far in past` |
|  500  |   ~50|
|  800  |   ~65|
| 1000  |   ~80|
| 1200  |   ~90|
| 1400  |   ~95|
| 1600  |   ~80|
| 1800  |   ~80|
| 2000 | ~78|

Greater ticks (which is what we really have in real life, if I correctly understand what tick is) produce greater load on CPU with IRQ being a bit more effective than polling on 500 — 800 ticks. 

On the other hand if there are a lot of gpio calls to be made — polling becomes more desirable method to use as it allows greater rates.

KevinOConnor, 2021-04-18:
> Best results are:
    2 stepper, 237 ticks, 422K
    1 stepper, 230 ticks, 217K

FYI, the one and two stepper benchmarks aren't particularly interesting as those results are primarily determined by the 2us step delay.  (The most we can possibly step a stepper with a 2us delay is 250K/s, but we can't actually reach more than ~200K/s in practice.)  The 3 stepper benchmark (or more) is needed to get real information on how well the code can issue steps.

Cheers,
-Kevin

tntclaus, 2021-04-18:
I just ran test g-code using irq + native gpio branch:

* 3  minute run
* speeds up to 300 mm/s
* acceleration 40K
* travels along x and y at 520 mm distance each.

The timer window is 1000 (not 5000 like last time) and no "stepper too far in past" errors.

>The 3 stepper benchmark (or more) is needed

Ok. Will prepare it a bit later.

KevinOConnor, 2021-04-18:
FYI, I extracted some of the optimizations from the work-linux-irq-20200607 branch and committed them to the master branch.  I've also rebased the work-linux-irq-20200607 branch on top of that new master branch.  The work-linux-irq-20200607 branch now just has the code to use signals.

In benchmarks, I now get similar results between master and work-linux-irq-20200607.  (Which makes sense, as signals may improve scheduling jitter, but shouldn't impact the benchmarks because there are always timers pending during a benchmark.)

-Kevin

tntclaus, 2021-04-18:
Did benchmark of this branch:
https://github.com/tntclaus/klipper/commits/work-rpi-gpio-20201008

Highest rate is **1017K** with **6 steppers**

| Steppers  | Ticks | Rate |
| -----------: | ------------: | ------: |
|  3 steppers  |  237 | 633K | 
|  4 steppers  |  247 |  810K |
|  5 steppers  |   267 | 936K |
|  6 steppers  |  295 | 1017K |
|  7 steppers  |  357 | 980K |

**3 steppers**
```
SET start_clock {clock+freq}
SET ticks 237

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0

reset_step_clock oid=2 clock={start_clocEval: SET start_clock 11417779484
k}
set_next_step_dir oid=2 dir=0
queue_step oid=2 interval={ticks} count=60000 add=0
set_next_step_dir oid=2 dir=1
queue_step oid=2 interval=3000 count=1 add=0

Eval: reset_step_clock oid=0 clock=11417779484
Eval: queue_step oid=0 interval=237 count=60000 add=0
Eval: reset_step_clock oid=1 clock=11417779484
Eval: queue_step oid=1 interval=237 count=60000 add=0
Eval: reset_step_clock oid=2 clock=11417779484
Eval: queue_step oid=2 interval=237 count=60000 add=0
219.337: stats count=71 sum=70229 sumsq=662248
224.417: stats count=56 sum=33754 sumsq=267569
ECHO Test result is: {"%.0fK" % (3. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 633K
```

**4 steppers**
```
SET start_clock {clock+freq}
SET ticks 247

reset_step_clock oid=0 clock={start_clock}
set_next_step_dir oid=0 dir=0
queue_step oid=0 interval={ticks} count=60000 add=0
set_next_step_dir oid=0 dir=1
queue_step oid=0 interval=3000 count=1 add=0

reset_step_clock oid=1 clock={start_clock}
set_next_step_dir oid=1 dir=0
queue_step oid=1 interval={ticks} count=60000 add=0
set_next_step_dir oid=1 dir=1
queue_step oid=1 interval=3000 count=1 add=0

reset_step_clock oid=2 clock={start_clock}
set_next_step_dir oid=2 dir=0
queue_step oid=2 interval={ticks} count=60000 add=0
set_next_step_dir oid=2 dir=1
[... 169 lines cut ...]
Eval: reset_step_clock oid=3 clock=21023307895
Eval: queue_step oid=3 interval=357 count=60000 add=0
Eval: reset_step_clock oid=4 clock=21023307895
Eval: queue_step oid=4 interval=357 count=60000 add=0
terval=3000 count=1 add=0

reset_step_clock oid=5 clock={start_clock}
set_next_step_dir oid=5 dir=0
queue_step oid=5 interval={ticks} count=60000 add=0
set_next_step_dir oid=5 dir=1
queue_step oid=5 interval=3000 count=1 add=0

reset_step_clock oid=6 clock={start_clock}
set_next_step_dir oid=6 dir=0
queue_step oid=6 interval={ticks} count=60000 add=0
set_next_step_dir oid=6 dir=1
queue_step oid=6 interval=3000 count=1 add=0
Eval: reset_step_clock oid=5 clock=21023307895
Eval: queue_step oid=5 interval=357 count=60000 add=0
Eval: reset_step_clock oid=6 clock=21023307895
Eval: queue_step oid=6 interval=357 count=60000 add=0
306.616: stats count=88 sum=84892 sumsq=586134
ECHO Test result is: {"%.0fK" % (7. * freq / ticks / 1000.)}
Eval: ECHO Test result is: 980K
```

tntclaus, 2021-04-18:
> FYI, I extracted some of the optimizations from the work-linux-irq-20200607 branch and committed them to the master branch. I've also rebased the work-linux-irq-20200607 branch on top of that new master branch. The work-linux-irq-20200607 branch now just has the code to use signals.

So does it mean I can create pull request for my [work-rpi-gpio-20201008](https://github.com/tntclaus/klipper/commits/work-rpi-gpio-20201008) branch so it could be merged to master? If so, can you please check if there are anything I should change in order to comply with how you are used to keep your code base?

KevinOConnor, 2021-04-18:
> Highest rate is 1017K with 6 steppers

Impressive.

> Does it mean I can create pull request for my work-rpi-gpio-20201008 branch so it could be merged to master?

Seems like useful functionality to me, so yes.

> If so, can you please check if there is anything I should change in order to comply with how you are used to keep your code base?

It's really hard for me to review code on other git trees.  If you can rebase your native gpio code to a single commit on top of the master branch, and create a PR with that, it will be much easier for me to review.

That said, I did see a bunch of #ifdefs added to the code.  I think it would be preferable to create a new file (eg, bcm1234_gpio.c) and change the src/linux/Makefile to include that file (instead of gpio.c) when using the rpi gpio code.

> Should I include these benchmarks in my pull request as well?

Updated benchmarks would be great, but I'd recommend committing the code first, then commit the docs update on top of that.

Cheers,
-Kevin

tntclaus, 2021-04-18:
Okay, will do that during next week.

tntclaus, 2021-04-19:
@KevinOConnor I have questions:

1. Should we still claim GPIO via
```
strncpy(req.consumer_label,GPIO_CONSUMER,sizeof(req.consumer_label) - 1);
    rv = ioctl(
        get_chip_fd(g.line->chipid),GPIO_GET_LINEHANDLE_IOCTL,&req);
```
so it would be visible and locked to other apps?

Argument for **NO**: there should be no other app controlling gpio on 3d printer, almost all of the required stuff can or even should be controlled via klipper and if one would require to launch separate app or script to control something through gpio he should take care of not using same pins as klipper already does. (Not fool-proof!)

Argument for **YES**: there possibly can be hazardous situations like simultaneous controlling from 2 different apps/scripts of, say, heater or spindle pwm pins with the risk of fire or trauma.

2. Do we need to keep `DECL_ENUMERATION_RANGE` for `gpiochipN`? AFAIK RPi has single gpio chip, seems for me there is no need in this code. 

What do you think?

tntclaus, 2021-04-22:
@KevinOConnor I filed pull request. At the last minute I removed RPi 4 menuconfig option as RPi 4 performance is twice worse than RPi3. 

I did following test:

# New RPi4 2GB version
# Blank OctoPi bistro
# Klipper from my dev branch
# Same benchmark as for RPi 3 — 2 times worse  performance.

Then I did following:

# Took another RPi 3 B (not the one I used before) and swapped SD card from RPi 4 to this new device.
# Booted up and reflashed klipper version fro RPi 3
# Run benchmarks and got same good results as I had on another RPi 3

Do you have any ideas why RPi 4 may have worse performance than RPi 3? That's odd as the gpio benchmarks I had seen this summer showed around 56 MHz gpio toggle rate.

tntclaus, 2021-04-22:
With RPi 4 it is possible to achieve somewhat close to RPi3 performance adding to `/boot/config` following:

```
over_voltage=6
arm_freq=2000
gpu_freq=520
```

For 6 and more steppers performance becomes better:

| Steppers | Ticks | Rate |
|---|---|---|
| 6 | 277 | 1083K | 
| 7 | 277 | 1264K | 

**But 1-5 steppers performance is worse.** 

This is weird as I was able to achieve as much as 65MHz GPIO toggle rate, which is almost  2 times greater than RPi 3 gpio toggle rate (36 MHz).

KevinOConnor, 2021-04-23:
> Should we still claim GPIO via

FWIW, I'd try to keep the code simple.

> Do you have any ideas why RPi 4 may have worse performance than RPi 3?
...
But 1-5 steppers performance is worse.

Once you get to ~200K steps per second on each stepper, it's no longer testing the gpio toggling, and is instead testing the code's ability to delay.  Minor changes in code layout can make for minor differences in delays.  The benchmark just isn't particularly meaningful if it's delay limited instead of cpu limited.

On some of the boards you'll see benchmarks with CONFIG_STEP_DELAY=0 .  This wont be a valid mode to use with real stepper motor drivers on an RPi, but that test might give a more stable result.

-Kevin

tntclaus, 2021-04-23:
> FWIW, I'd try to keep the code simple.

I kept the ioctl/sysfs part of code as it is still used by `gpio_generic.c` and shared it with `gpio_bcm.c`. So we have best of both worlds: the code is still simple, is not duplicated and safety is preserved at the level we are used to. 

The safety is very important as because one of the greater values of Klipper for me is that it's fool proof: the firmware already saved my ass couple times from starting fire and/or melting printer parts at my early days with 3d printing last summer. I think this is good opportunity to say "Thank you, Kevin" ;)

> instead testing the code's ability to delay

I understand, but code is the same at RPI 3 and RPi 4 and the result of RPi4 is worse. My expectation was for it to be the same for small amount of steppers (I was completely wrong) and better for 6+ steppers  (this is somewhat what happened).

Couple of days ago I've got an idea to make a timer which is bound to RPi hardware PWM (the code that resides in DMA).

closed 2021-06-06 by github-actions[bot]