Klipper3d/klipper #3389: Timer too close,

mistrjirka, 2020-10-01:
Hi I have been trying to print, when randomly this error ocours: 
MCU 'mcu' shutdown: Timer too close
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
I read this is speed or priority issue, but it did not seem it was really that. I have odroid C1+ and it has 4 core cpu, that is almost not utilized when klipper and octoprint are at full use, but after I write this issue I will try to make it better with priority and print again the same gcode. But I wanted to post this before I loose log files, it also was not that much fast print and my printer is cartesian so it is not that hard to compute the movements. It happend first time ever, but before I sometimes had a issue with No next step, that was solved by updating klipper so this can be a new bug. Here are gcodes, logs. 
Important to mension the gcode is cutted previouse gcode because it failed two times in the same print. 
[logsgcodes.zip](https://github.com/KevinOConnor/klipper/files/5312775/logsgcodes.zip)
The priorities are 
[image]

mistrjirka, 2020-10-01:
Update: I git pulled new version and I will try to save the print that the klipper was failing on. I will update for results

KevinOConnor, 2020-10-02:
See #3331 / #3316 - as you've indicated, updating to the latest code is the next step.

-Kevin

closed 2020-10-31 by github-actions[bot]