Klipper3d/klipper #3390: Move out of range

moedope, 2020-10-01:
so ive been getting this error and i have no idea why as my bed size is 300x300 and the piece is 140mm on x
ive issued g1 commands on x y to see if its going over boundries and its not. attach my log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5314658/klippy.log)

jakep82, 2020-10-01:
You need to define a position_min in your stepper_z section that is below zero.  Your bed mesh shows a low point of about -0.5mm.  I would also suggest trying to fix your bed level.  Your mesh shows a difference of about 1.25mm between your high spot and your low spot.  That's going to cause major issues with your first layer, and bed mesh won't entirely solve that.

moedope, 2020-10-03:
thank you, issue fixed, set z off set to  a - value in z stepper section

closed 2020-10-03 by moedope