Klipper3d/klipper #3391: Markforged Kinematics

AndiBl, 2020-10-02:
Hi,
would it be possible to implement the Markforged Kinematics, like in the newest Marlin Version?
https://github.com/MarlinFirmware/Marlin/pull/19235
Thanks!

KevinOConnor, 2020-10-02:
I'm sure it's possible.  An interested developer would need to take that on and submit the resulting code.  It looks like it would only require host Python code changes.

-Kevin

Tircown, 2020-11-10:
Hi @AndiBl,

I am working on the implementation of markforged kinematic for klipper. At this point it works both for one carriage and dual-carriages. I continue to work on lesser known modes for idex: duplication and mirroring.
You can see where I am at on my test rig: youtube.com/watch?v=esPHxrPIq_o

I have few questions for @KevinOConnor 
1.  Since kinematic's formula for markforged are not the same depending on where are endstops I would need to add a parameter to invert movement. This parameter should be in printer section I guess? I don't know exactly when to invert but my test rig is made for that. Atm these are empirical observations: with y endstop at max I need to invert (like in the video), with y endstop at min, I dont need to.
2. I can not manage easily the dual carriage on y axis so I plan to forbidden it. It creates a particular case between cartesian and markforged. Is that a problem for you? Furthermore I can't see any real cas use.
3. I can easily add markforged kinematic on an xz gantry. Should I add a parameter in printer section or create a second kinematic as it is done for corexy/corexz?
4. Are offset X and Y taken in account for triggering the limits?

Best Regards,
Tircown

KevinOConnor, 2020-11-10:
> Since kinematic's formula for markforged are not the same depending on where are endstops I would need to add a parameter to invert movement. 

I'm confused why the endstop position would change the kinematics.  Normally, the direction to move during homing is determined by the `position_endstop` parameter (see also `homing_positive_dir` in config/example.cfg).  Or, are you saying that same "markforged" machines control the x axis using x+y kinematics and some use x-y kinematics?

> I can not manage easily the dual carriage on y axis so I plan to forbidden it. It creates a particular case between cartesian and markforged. Is that a problem for you?

That is fine with me.

> I can easily add markforged kinematic on an xz gantry. Should I add a parameter in printer section or create a second kinematic as it is done for corexy/corexz?

I would only add corexz support if at least one person can verify it works on a real machine.

> Are offset X and Y taken in account for triggering the limits?

I don't understand the question.  Perhaps it is related to `position_endstop` - in which case take a look at the description in config/example.cfg .

Cheers,
-Kevin

Tircown, 2020-11-10:
Thank you @KevinOConnor .

>Or, are you saying that some "markforged" machines control the x axis using x+y kinematics and some use x-y kinematics?

Yes, the calc_position for y is the cartesian one, always. For x, it is x-y when I set the stepper_y:endstop_position to 0 and x+y when I set my stepper_y:endstop_position to the max, 200mm.

>I would only add corexz support if at least one person can verify it works on a real machine.

I guess you mean MarkforgedXZ, as CoreXZ already exists in Klipper. To give a try I just need a simple config change on my test rig. This kinematic could be used for i3 type printers close to the design of a Voron/Switchwire but in IDEX.

>Are offset X and Y taken in account for triggering the limits?

Ok I clarify my thoughts. First, by offsets I mean the values put in the SET_GCODE_OFFSET command. For duplication and mirroring mode the moving range should be modified to avoid carriages collision. Typicaly for theses modes one doesn't have to set offsets. But if there is an offset X set, it would lead to an error to simply divide the original range by 2. So, how the limits class deals with these offsets?

Fabrice

KevinOConnor, 2020-11-13:
> For x, it is x-y when I set the stepper_y:endstop_position to 0 and x+y when I set my stepper_y:endstop_position to the max, 200mm.

Okay - if the two carriages have different kinematics (x+y vs x-y) then that will need to be configured somewhere.  I would configure it directly; I would not make it dependent on the endstop postion.

> First, by offsets I mean the values put in the SET_GCODE_OFFSET command.

The SET_GCODE_OFFSET only impacts g-code commands - it basically alters later g-code commands to add the given offset to it.  It does not have any other impact nor is it configured anywhere other than via the SET_GCODE_OFFSET command itself.

-Kevin

closed 2020-12-12 by github-actions[bot]