Klipper3d/klipper #3393: Protocol Error after update

joshschools, 2020-10-03:
Just updated Klipper and receive the following protocol error:

`Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 573, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 512, in _send_config
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 251, in _build_config
    cq=cmd_queue)
  File "/home/pi/klipper/klippy/mcu.py", line 669, in lookup_command
    return CommandWrapper(self._serial, msgformat, cq)
  File "/home/pi/klipper/klippy/mcu.py", line 386, in __init__
    self._cmd = serial.get_msgparser().lookup_command(msgformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 303, in lookup_command
    msgformat, mp.msgformat))
error: Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u
No build file /home/pi/klipper/klippy/../.config
No build file /home/pi/klipper/klippy/../out/klipper.dict
No build file /home/pi/klipper/klippy/../out/klipper.elf
`
[klippy.log](https://github.com/KevinOConnor/klipper/files/5321844/klippy.log)

ShohninDmitriy, 2020-10-03:
> Just updated Klipper and receive the following protocol error:

Recompile the firmware and re-flash the controller

joshschools, 2020-10-03:
That worked!

closed 2020-10-03 by joshschools