Klipper3d/klipper #3394: moving /testing one stepper.

rewolff, 2020-10-03:
My Delta 3D printer refuses to move until I hit "home". Understandable. 

There is an issue with one of the steppers. I don't know what yet. When I hit "home" it doesn't move. To debug this I'd like to be able to say: 

DEBUG_MOVE_TOWER A -12.34

When things are not moving, things are awkward while homing. The head moves outside the envelope (when it starts low) and when I swap connectors to see if it is the motor or the driver things hit endstops hard and don't stop when they are supposed to if I test by hitting "home"... So hitting home all the time is not a good idea. I much prefer to do something more subtle....

rewolff, 2020-10-03:
Totally irrelevant but the bot closes issues that don't have a log. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5321980/klippy.log)

KevinOConnor, 2020-10-03:
I use the STEPPER_BUZZ command - see: https://www.klipper3d.org/Config_checks.html .  There are also other options described in the FAQ: https://www.klipper3d.org/FAQ.html#why-cant-i-move-the-stepper-before-homing-the-printer

-Kevin

rewolff, 2020-10-03:
Aaargh. Found it myself already. Didn't use the right keywords when searching. I have enabled force_move and so far I've saved myself some work: Turns out the towers are incorrectly tagged, so the cable that "looks OK but apparently doesnt' work" was of a tower that did work. (To verify something: i'd swapped the one that I thought didn't work and the one that actually didn't work and got the expected effects.... aargh!)

closed 2020-10-03 by rewolff