Klipper3d/klipper #3396: Motors not holding current

acastles91, 2020-10-04:
I am using klipper on a pen plotter with an airbrush mounted on a wall. It is working fine and I'm getting used to the system and its parameters, but one thing that I haven't been able to solve is that my Y axis (which, since the printer is mounted on the wall, is vertical, parallel to the wall) stops holding the weight while printing and the carriage drops some centimeters every once in a while. It looks like a sort of sleep function, I don't know. The y axis ends up missing some steps but the print continues and the axis keeps moving. I am using TMC5160 and I disabled stealthchop, on an SKR 1.4. Does it have to do with the hold current? I'm using Nema 23 motors.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5323332/klippy.log)

theophile, 2020-10-05:
Your log indicates your Klipper version is a few months old so I'd try updating and seeing if that helps.

If not, you might check on the Standstill Detection feature of the TMC5160 driver. It looks like it has a mode that will automatically power down the stepper after some configured delay following the driver's detection that the motor is standing still. Since your Y axis moves vertically like the Z axis typically does on a cartesian printer, you probably don't want it doing that. I don't know offhand how to disable it, but it appears that increasing the `driver_tpowerdown` value will increase "the delay time after stand still (stst) of the motor to motor current power down." (https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC5160A_Datasheet_Rev1.14.pdf , page 38). Right now it's set to 10 seconds. If you're noticing the carriage drop after about 10 seconds of inactivity, that might be your culprit.

KevinOConnor, 2020-10-07:
I don't know why that would occur - it sounds like a stepper driver or hardware issue.  I'd try with stealthchop disabled (set `stealthchop_threshold` to zero to disable stealthchop).

-Kevin

acastles91, 2020-10-20:
Thank you for your help. I tried changing the values of `driver_IHOLDDELAY` and `driver_TPOWERDOWN` to the maximum settings but I'm still getting the same problem. I also defined `stealthchop_threshold = 0`. I don't know about it being a stepper driver issue, since the printing seems to be consistent for the most part. It just drops at some point a bit keeps on printing without stopping.

This is the new klippy log, is there a sleep mode or something similar? I remember Marlin having something like that, and since it is only happening in the Y axis, which is the one that moves slower, maybe it has something to do with that.

Any help is appreciated. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5409399/klippy.log)

KevinOConnor, 2020-10-22:
It doesn't sound related to Klipper.  It could be related to overheating of the steppers/drivers and/or voltage drop in the power supply.  You may have better luck asking in a general 3d printing or CNC forum.

-Kevin

closed 2020-11-20 by github-actions[bot]