Klipper3d/klipper #3397: Add ability to ignore first probe value

Makhaira, 2020-10-04:
I have seen a number of reports across different printers and different probes where the first probe result of a "set" is inaccurate compare to the rest of the set of results. It would be great if there was a config variable in klipper where you could choose to ignore the first result in a probe.

E.G.

probe returns result of 0.2, 0.19, 0.19, 0.19, 0.19. 

I am asking for klipper to ignore the first result in any subsequent calculations, the dataset becomes just 0.19, 0.19, 0.19, 0.19. This has the potential to remove the number of repeat probes required as there is confidence the first result is ignored.

Sineos, 2020-10-05:
You can use the `median` function instead of `average`. IMO the best approach is to use an even number of probes, as the median function will average around the middle.

For example:
0.18, 0.19, 0.21, 0.22
will discard 0.18 and 0.22 and average 0.19 and 0.21

Makhaira, 2020-10-07:
Thanks Sineos. I realise that is one solution, however many probe solutions
now have such orecision/repeatability that you don't even need that many
samples. I know that's true for the Omron inductive probes, and apparently
for some of the microswitch based ones too.

I realise this isn't a bug or a fault, but it would be a neat improvement
to speed up probing.

Makhaira, 2020-10-29:
Will use median on even number probe

closed 2020-10-29 by Makhaira