Klipper3d/klipper #3398: "Rescheduled timer in the past" during PA tune

Sineos, 2020-10-04:
Today I updated to https://github.com/KevinOConnor/klipper/commit/0bf0cb5b547af5509cb73c567c0214e408134e1d and also flashed the SKR 1.4 Turbo MCU new.
Trying to run a PA tune ended up in `Rescheduled timer in the past`
- The error already happens during the PRINT_START macro
- With the very same setting I did countless PA tunes before the update / reflash
- I tried reducing microstepping from 64 to 32 (although 64 worked before for PA tune and regular printing)
- Tried repeating the build process for both linux and LPC MCU after `make clean` / `make distclean`
- A regular print works

Attached: 
- klippy.log: PA tune
- klippy2.log: Same settings with a regular print

[klippy2.log](https://github.com/KevinOConnor/klipper/files/5323760/klippy2.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5323761/klippy.log)

dmbutyugin, 2020-10-05:
From the logs:
> extruder: pressure_advance: 30.001500

It sounds like in `TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=30 FACTOR=.005` command `START=30` is a mistake.

Sineos, 2020-10-06:
Ouch, that is embarrassing.
Of course this should have read 0.3 and it totally slipped my attention. Thanks for spotting the obvious.

closed 2020-10-06 by Sineos