Klipper3d/klipper #3399: Klipper Printer not ready

ChrisBierings, 2020-10-04:
I just installed klipper for a few days and was configuring it and tuning my bltouch settings but when they where ready and i wanted to do a test print it spits out a error box with the text: "Your printer's firmware reported an error. Due to that the ongoing print job will be cancelled. Reported error: Printer is not ready" and in the terminal is says this after the bed mesh stuff: "
Recv: ok
Send: N10 G92 E0.0*104
Recv: // Klipper state: Disconnect
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Changing monitoring state from "Printing" to "Cancelling"
Send: N11 M104 S205.0*76
Recv: ok
Recv: // Printer is not ready
Send: N12 G1 X0.1 Y20 Z0.3 F5000.0*29
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Send: N13 M108*24
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: N14 M84*42
Recv: // Printer is not ready
Send: N15 M104 T0 S0*21
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: N16 M108*29
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Send: N17 M140 S0*83
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: N18 M106 S0*94
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: N19 M108*18
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Changing monitoring state from "Cancelling" to "Operational"
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Send: M84
Recv: ok
Send: M108
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M104 T0 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M140 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Send: M108
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Send: M106 S0
Recv: ok
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Send: M108
Recv: ok
[...]
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M84
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M108
[...]
Send: M104 T0 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M140 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M108
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Send: M106 S0
Recv: ok
Recv: // Printer is not ready
Send: M84
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M104 T0 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M140 S0
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Send: M106 S0
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Printer is not ready
Send: M84
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Send: M104 T0 S0
Recv: // Klipper state: Ready
Recv: ok
Recv: ok
Recv: ok
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
Send: M84
Recv: ok
Send: M104 T0 S0
Recv: ok
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
Send: M84
Recv: ok
Send: M104 T0 S0
Recv: ok
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
Send: M84
Recv: ok
Send: M104 T0 S0
Recv: ok
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
" 
so i did a lot of research but for almost everyone the fix is to change the serial under mcu in their config file but for me that didn't work so i tryed to reflash klipper 3 times but it still doesn't work so hopefully someone can help me :)

Klipper log file: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5324413/klippy.log)

-Chris

Arksine, 2020-10-04:
Does the gcode file you are attempting to print have a call to `SAVE_CONFIG` in it?  It appears so, and that restarts the firmware.

ChrisBierings, 2020-10-04:
> Does the gcode file you are attempting to print have a call to `SAVE_CONFIG` in it? It appears so, and that restarts the host.

no but i do have this in my printer.cfg: 
"[gcode_macro START_PRINT]
gcode:
    G28; Home all axes    
    BED_MESH_CALIBRATE    
    G92 E0; Reset Extruder
    G1 Z5.0 F3000; Move Z Axis up little to prevent scratching of Heat Bed
    G1 X0.1 Y20 Z0.3 F5000.0; Move to start position
    G1 X0.1 Y200.0 Z0.3 F1500.0 E15; Draw the first line
    G1 X0.4 Y200.0 Z0.3 F5000.0; Move to side a little
    G1 X0.4 Y20 Z0.3 F1500.0 E30; Draw the second line
    G1 E25; retract 5mm
    G92 E0; Reset Extruder
    G4 S1
    G1 Z10 F3000; Move Z Axis up little to prevent scratching of Heat Bed
"
and if i need to do that where should i place it (after which line)?

Arksine, 2020-10-04:
Ah, it is in your G29 macro. 
```
[gcode_macro G29]
gcode = 
	G1 Z10 F600
	G4 P200
	G1 Z10 F600
	G4 P100
	BED_MESH_CALIBRATE
	BED_MESH_OUTPUT
	SAVE_CONFIG
```
If G29 is being called from the print, it executes the above macro, which restarts the Klippy Host.  You will want to remove SAVE_CONFIG so that this does not happen.

ChrisBierings, 2020-10-04:
> Ah, it is in your G29 macro.
>
> ```
> [gcode_macro G29]
> gcode =
> 	G1 Z10 F600
> 	G4 P200
> 	G1 Z10 F600
> 	G4 P100
> 	BED_MESH_CALIBRATE
> 	BED_MESH_OUTPUT
> 	SAVE_CONFIG
> ```
>
> If G29 is being called from the print, [...]

Thxx it works now :)

closed 2020-10-04 by ChrisBierings