Klipper3d/klipper #3400: [FR] Firmware Retraction Settings Per Extruder

Block137, 2020-10-05:
Allow different firmware retraction settings for each extruder for multi-extruder printers.

The config section doesn't specify which extruder so I assumed it applies to all extruders which isn't optimal when printing with different material and/or different extruder design.

KevinOConnor, 2020-10-07:
It's recommended to define a macro to use when switching extruders (eg, `T0`).  If it is desired to alter the firmware_retraction for each extruder then the best way to do this is to update that macro to issue a `SET_RETRACTION` command.

-Kevin

closed 2020-10-22 by Block137

Block137, 2020-12-23:
For others, here's my current solution: redefined firmware retraction so I can also include Z-hop. No problem so far

```
[gcode_macro G10]
variable_retracted_t0: 0
variable_retracted_t1: 0
variable_retracted_t2: 0
variable_retract_t0: 0.5
variable_retract_t1: 11.0
variable_retract_t2: 7.0
variable_retract_spd_t0: 60
variable_retract_spd_t1: 60
variable_retract_spd_t2: 60
variable_z_t0: 0.25
variable_z_t1: 0.3
variable_z_t2: 0.3
variable_z_spd_t0: 15
variable_z_spd_t1: 15
variable_z_spd_t2: 15
gcode:
    SAVE_GCODE_STATE
    {% if printer.toolhead.extruder == 'extruder' %}
        {% if retracted_t0 == 0 %}
            M83
            G1 E-{retract_t0} F{retract_spd_t0 * 60}
            RESTORE_GCODE_STATE MOVE=0
            SET_GCODE_OFFSET Z_ADJUST={z_t0} MOVE=1 MOVE_SPEED={z_spd_t0}
            SET_GCODE_VARIABLE MACRO=G10 VARIABLE=retracted_t0 VALUE=1
            SET_GCODE_VARIABLE MACRO=G11 VARIABLE=retracted_t0 VALUE=1
        {% endif %}
    {% elif printer.toolhead.extruder == 'extruder1' %}
        {% if retracted_t1 == 0 %}
            M83
            G1 E-{retract_t1} F{retract_spd_t1 * 60}
            RESTORE_GCODE_STATE MOVE=0
            SET_GCODE_OFFSET Z_ADJUST={z_t1} MOVE=1 MOVE_SPEED={z_spd_t1}
            SET_GCODE_VARIABLE MACRO=G10 VARIABLE=retracted_t1 VALUE=1
            SET_GCODE_VARIABLE MACRO=G11 VARIABLE=retracted_t1 VALUE=1
        {% endif %}
    {% elif printer.toolhead.extruder == 'extruder2' %}
        {% if retracted_t2 == 0 %}
            M83
            G1 E-{retract_t2} F{retract_spd_t2 * 60}
            RESTORE_GCODE_STATE MOVE=0
            SET_GCODE_OFFSET Z_ADJUST={z_t2} MOVE=1 MOVE_SPEED={z_spd_t2}
            SET_GCODE_VARIABLE MACRO=G10 VARIABLE=retracted_t2 VALUE=1
            SET_GCODE_VARIABLE MACRO=G11 VARIABLE=retracted_t2 VALUE=1
        {% endif %}
    {% else %}
        
    {% endif %}
```