Klipper3d/klipper #3404: 'driver_HEND' in section 'tmc2209 stepper_x' must have minimum of 0

kakou-fr, 2020-10-06:
I am to reduce the noice/loud due to 0.9 stepper and TMC2209 on a Core-XY.

BLV and https://github.com/MarlinFirmware/Marlin/issues/18542 recommand to use : 
```
#define CHOPPER_09STEP_24V   { 3, -1, 5 }
{ <off_time[1..15]>, <hysteresis_end[-3..12]>, hysteresis_start[1..8] }
```
but if I define : 
```
driver_TOFF: 3
driver_HEND: -1
driver_HSTRT: 5
```
II have this error : 
```
Error: Option 'driver_HEND' in section 'tmc2209 stepper_x' must have minimum of 0
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/5335595/klippy.log)

Hywelmartin, 2020-10-06:
perhaps this???
[image]
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V105.pdf

kakou-fr, 2020-10-06:
so marlin use 
-3..12 and klipper 0..15 for hend
1..8 and klipper 0..7 for HSTRT

i will test with 
```
driver_TOFF: 3
driver_HEND: 2
driver_HSTRT: 4
```

but finally, i desactivated stealchop and tested only with spreacycle. It's very quiet and no loud at 9000mm/min. tested until 400mm/s without strange sound. Looks like tmc2209+stealchop+COREXY is the problem

kakou-fr, 2020-10-07:
ok it works silence now until 40000mm/min  (max tested) in spreadcycle only mode at 32mstep and  0.9 E3D motor.

closed 2020-10-07 by kakou-fr