Klipper3d/klipper #3405: SKR Pro display doesn't work, hotend PID integrator windup, firmware retract disregards retract speed

DroneMang, 2020-10-06:
Hi,

I have an SKR Pro with a BIGTREETECH TFT24 V1.1 Touch Screen Display 12864LCD, hooked up in the emulation mode using the EXP1 and 2 connectors so it should work just like a RepRapDiscount 128x64 Full Graphic Smart Controller which is what I have defined.  I tried the code block from the lcd sample document which is commented out, and some other values from another person with an SKR Pro, neither work.

Link to Display:
https://www.aliexpress.com/item/33048619460.html?spm=a2g0s.9042311.0.0.27424c4dQRD6IT

Also, when my cooling fan turns on after the first 3 layers, the print fails due to hotend temp dropping too much.  I have set the timeout to 360 to help this.  I think the integrator in the PID has wound up and so it takes it a long time to adjust to the new setpoint.  Perhaps you could reset the PID but keep same calibration values if the cooling fan speed changes?  I have a large bed with the FT-6 so large prints cause this error.

I am using firmware retract and pressure advance.  With a volcano hotend I found that setting the retract and unretract speeds very high (I have had 600 mm/s set in marlin) virtually eliminated ooze or gaps when starting a new line as the hotend moves immediately to the new point.  I have tried setting high retract and unretract speeds in klipper, but as far as I can tell, no change to speed happens.  Is there a max speed that is allowed?

Lastly, my log file is not working today, I can't find one anywhere, either in octoklipper or with filezilla, so this is just my config.  The log file was working yesterday though so I do not understand what is happening there.  This occured after adding in a section for filament runout sensors, M600, and the current uncommented display block.

[klippy (2).zip](https://github.com/KevinOConnor/klipper/files/5336043/klippy.2.zip)

jakep82, 2020-10-06:
This screen does not work properly with Klipper with stock BTT firmware.  Some people report limited success using custom screen firmware, but it sounds like issues still exist.

https://github.com/KevinOConnor/klipper/issues/2231
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/705

KevinOConnor, 2020-10-07:
You'll need to find and attach the klipper log in order for us to assist.  If the klipper log file is not in /tmp/klipper.log then it may indicate a problem with your SD card - you may need to reinstall the OctoPi image.

-Kevin

DroneMang, 2020-10-07:
The TFT24 display doesn't appear to work with Klipper.  I just tried this with the precompiled firmware from BTT and by compiling it, with the unified menu and standard, and nothing shows up other than LCD12864 simulator.  I have an SKR pro.

I got the log file working again by putting in the standard lcd block.  Interesting that having the lcd defined wrong causes the log to fail but everything else works.

closed 2020-11-02 by DroneMang