Klipper3d/klipper #3406: Z_Tilt_adjust not working properly

Roycinger, 2020-10-07:
Hi folks, 
I'm seeking advice on my Z_Tilt.
I've set up a [z_zilt] section in my config like this on a i3 machine:
```
[z_tilt]
speed: 100
z_positions: 
 -62,100
 268,100
points:
 10,100 # left probe
 168.2,100 # right probe
retries: 3
retry_tolerance: 0.04
```

I've measured the positions of the left and right Z leadscrews (relative to the nozzle) as best as I can.
While doing the Z_TILT_ADJUST, I see the two Z motors adjusting the x axis.
However, when I check for same nozzle heights on the left and right side of the bed, on the left, the nozzle is touching the bed, while on the right, the nozzle has a distance of about 0.2mm to the bed.
My first thought was that the leadscrew position are off. But changing the position values in the config didn't yield any noticable result.
Is there a point I'm missing here? Thanks for any help!
[klippy.log](https://github.com/KevinOConnor/klipper/files/5339635/klippy.log)

KevinOConnor, 2020-10-13:
It's difficult to say why that would be - as far as we know, the code works as intended.  It may also be a "location bias" of the probe - see docs/Probe_Calibrate.md for a description of that.

-Kevin

Roycinger, 2020-10-13:
Thanks.
In the meantime, I tried removing the mesh beforehand with the same result. I'll go through the section mentioned above and report back

marcus-wu, 2020-10-17:
@Roycinger curious what you're using for probe. I noticed if using BLTouch & stow_on_each_sample: False, Z_TILT_ADJUST never adjusts the motor. But if you leave default. ie, comment out stow_on_each_sample: False. It'll work fine

Roycinger, 2020-10-17:
@marcus-wu I'm using a BLTouch.
Don't get me wrong. The adjustment works, it's just not accurate enough for me. I still need to evaluate Kevin's comment.

Roycinger, 2020-11-08:
Issue doesnt seem to resolved via software

closed 2020-11-08 by Roycinger