Klipper3d/klipper #3409: Z_offset is ignored

cvioan, 2020-10-08:
[printer.zip](https://github.com/KevinOConnor/klipper/files/5348292/printer.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5348293/klippy.zip)

Ender 3 with skr v1.3 and bltouch. 
Updated to latest version of klipper.

Set the BLTOUCH z_offset with probe_calibrate.

When homing and going to 0.0 on Z the position is correct as set.
When it starts printing the position is always to close, I need to always add a 0.900 offset to Z in order for the printer to be able to print.

Changing the z offset manually (not in the current printer.cfg), by adjusting with 0.900 results in the same behaviour as without adjusting, always need to add the offset manually or in the gcode.

Arksine, 2020-10-08:
This is likely because you have a `relative_reference_index` configured in `bed_mesh`.  This setting should only be used when homing z with an endstop, as it overrides the offset applied to the mesh.

cvioan, 2020-10-08:
I will take it out, but that line was added today in an effort to see if it helps. The issue is reproducing since 2 days ago, when I have updated to the latest version at that date.

Previously I have used a branch by Arksine from more than a month ago which was the only way to make mainsail work. 

I forgot that I have made a clean install on Tuesday (6th October) only using the existing config, and updating as needed in order to make Klaipeda, moonraker and mainsail work. I will look in the commits of the past month to see if something changed regarding this issue.

For now I have the offset of 0.9 added as Goode in the starting script

cvioan, 2020-10-09:
Reading through other issues, seems related to [#3319](https://github.com/KevinOConnor/klipper/issues/3319)

Arksine, 2020-10-10:
Understood, however as with #3319 I need logging that clearly indicates the issue as stated.  If you are able to reproduce this issue it would be helpful to execute `GET_POSITION` prior to applying the gcode offset, then upload the resulting log.  If bed_mesh is the culprit here then it should clearly be indicated in the positional information.

cvioan, 2020-10-10:
I will attach the logs as soon as possible. I am out of town for the moment and will reply as soon as I get back. Thank you for your support

cvioan, 2020-10-12:
Ran the following gcode file

```
G28
G1 X30.0 Y30.0 Z5.0 F3000
GET_POSITION
BED_MESH_CALIBRATE
M400
G1 X30.0 Y30.0 Z5.0 F3000
M400
GET_POSITION
G1 X130.0 Y30.0 Z5.0 F3000
M400
GET_POSITION
```

and resulted in [klippy12102020.zip](https://github.com/KevinOConnor/klipper/files/5365610/klippy12102020.zip)

technoo10201, 2020-10-22:
Hello @cvioan Got the same issue here... My z_offset is ignored and my nozzle stays too close to the bed..

Arksine, 2020-10-24:
FWIW, Kevin pushed a potential fix for this issue.

technoo10201, 2020-10-24:
> FWIW, Kevin pushed a potential fix for this issue.

Hi, thanks for the info ! Do you have any link of the fix ?

cvioan, 2020-10-24:
It seems like it as far as I saw, still testing in order to confirm

cvioan, 2020-10-26:
I ca-n conform it is solved for me.

closed 2020-10-26 by cvioan