Klipper3d/klipper #3415: Delta with Probe: delta_calibrate  !! horizontal_move_z can't be less than probe's z_offset (Anycubic Kossel)

anjoku, 2020-10-10:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5358669/klippy.log)

Hello, I love klipper and have it installed on an OrangePi Lite.
Printer: Anycubic Kossel Linear with Sample

Automatic Probe works perfectly and is very accurate.
The probe offset is: z_offset = 16.075

But when I start DELTA_CALIBRATE, the error always comes:

!! horizontal_move_z can't be less than probe's z_offset

I have entered postive and negative values for horizontal_move_z: but the error remains.
Also larger than probe's  offset ie. +20 or -20 or +50 or -50.

I have read a lot, but I can't get any further.

Best regards

Andre

Hywelmartin, 2020-10-10:
Horizontal move in delta calibration section is normally 5
Add it  and set >offset

https://github.com/KevinOConnor/klipper/blob/930317fa2a40a4424ce0a65bb02f32e6472eeba0/config/example-delta.cfg#L121

anjoku, 2020-10-10:
hi, ok normaly 5.
the Offset of my probe is: z_offset = 16.075

i set the horizontal_move_z to 20.  20 is more than my probe Offset. i think that is ok.

but it don't work.

Hywelmartin, 2020-10-11:
In  add  horizontal_move_z: 20
Like

[delta_calinrate] 
...
....
horizontal_move_z: 20
...

anjoku, 2020-10-11:
Hello,

Many Thanks. I entered at [delta_calibrate] and it works.

I had thought that it would only be entered in the [bed_mesh] section.

Many Thanks

Andre

anjoku, 2020-10-11:
it works now

closed 2020-10-11 by anjoku