Klipper3d/klipper #3416: CoreXY Speed at 45degrees?

changedsoul, 2020-10-10:
I have a coreXY configuration with TMC5160 drivers. I noticed my motors switching in and out of StealthChop while it was doing a infill pattern at a 45deg. The programmed speed was 80mm/s, and I have "stealthchop_threshold: 100" for both X and Y drivers.

In a CoreXY config, only one motor moves when going at a 45 deg so I am confused at why it would be switching in and out of stealthChop if my threshold was at 100.
If it were moving only in the Y or X Axis, I would be expecting both motors to be moving slower than programed speed since how the coreXY works. So in a 45 deg I dont understand why its switching out before it reaches its threshold.

Can someone help shed some light on this?

changedsoul, 2020-10-10:
klipper.log
[klippy.log](https://github.com/KevinOConnor/klipper/files/5360211/klippy.log)

KevinOConnor, 2020-10-12:
> The programmed speed was 80mm/s, and I have "stealthchop_threshold: 100" for both X and Y drivers.

Toolhead speeds are relative to the print.  The stealthchop_threshold is relative to the speed of the stepper motor itself.  If you requested a diagonal move at 80mm/s on corexy, the stepper would move at 80*sqrt(2) ~= 113.137mm/s to accomplish the move at the requested speed.

Separately, I recommend setting stealthchop_threshold to either zero (to always use spreadcycle) or 999999 (to always use stealthchop).

-Kevin

changedsoul, 2020-10-12:
Thanks for the response. I am still a bit confused. If having one motor run moving the machine at a 45 causes the motor to run Speed*SQRT(2), then if both are running casing it to move parallel to X or Y, is this situation making the motors move ya programmed speed?

Also, I have heard the same of keeping either in stealth spread, but I have found I get the best of both worlds if my threshold is around 100ish. 
Higher and the switch over causes layer shifts, set to high and I cannot achieve fast rapids. I have been printing for a while now with threshold at 100 and even though I hear the clank switch over, I have had. I layer shifts. I don’t print faster than 80 usually, so if I set my threshold at about 120, that should cover my print being quiet and switch over to spreadcycle for the rapids.

closed 2020-10-16 by changedsoul