Klipper3d/klipper #3417: Part cooling fan always run at 100%

fkotey, 2020-10-11:
I have installed klipper on my ender 3 with skr mini e3 v1.2. Everything seems to be working. However, the part cooling fan is always running at 100%.

In order to make sure the board is not the culprit here, I flashed the board with Marlin 2.0 and the part cooling fan is working as it should. Is there anything I am missing? I have connected the part cooling fan to Fan0 on the board which is right connection to use. Thanks for the help.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5360622/klippy.log)

KevinOConnor, 2020-10-13:
That's certainly unusual.  If you comment out the fan section entirely and do a full power cycle of the board, does the fan still come on?

-Kevin

fkotey, 2020-10-13:
> 
> 
> That's certainly unusual. If you comment out the fan section entirely and do a full power cycle of the board, does the fan still come on?
> 
> -Kevin

No, the fan does not come when I do what you asked. Thanks

KevinOConnor, 2020-10-22:
That generally indicates something has commanded the fan on.  If you issue a `M107` command, does the fan turn off?

-Kevin

closed 2020-11-13 by fkotey