Klipper3d/klipper #3418: bed_mesh_calibrate fails with precision piezo

copardo, 2020-10-11:
Hey guys,

I need some help. I am stuck in the bed_mesh_calibrate by using the precision piezo. z-homing works fine, also the first sampl

e point of the bed calibration. At the next point, klipper says "#Recv: !! Failed to home probe: Timeout during homing". I am wondering because everything worked fine so far.
This I already did:
-I aligned everything to get rid of tilts being leftover
- activated samples_tolerance  and played around with the values
- activated samples_tolerance_retries 

I attached the logfile also containing my config config. I am using the hypercube, tmc2130. I hope anyone has an idea for me.

Thanks

Dan

Hywelmartin, 2020-10-11:
```
[probe]
...
activate_gcode:
         G4 P500
```

copardo, 2020-10-11:
> ```
> [probe]
> ...
> activate_gcode:
>          G4 P500
> ```

edited, but behavior didn't change....

jakep82, 2020-10-12:
You should do a probe calibration before trying to create a mesh.  A piezo will require a negative Z offset value in your probe section, and you'll also want to add a negative position_min value in your stepper_z section.  You're bed looks pretty out of level, and it's probably trying to probe a spot below where it homed, and timing out when it doesn't trigger when it reaches zero.

copardo, 2020-10-15:
this works, thanks! But can you tell me the relation between position_min and samples_tolerance?

Hywelmartin, 2020-10-15:
position_min is how low the axes is allowed to go and samples_tolerance is how much difference that is allowed between probes

closed 2020-11-14 by github-actions[bot]