Klipper3d/klipper #3421: Move Extruder and MANUAL_STEPPER in parallel

modmt, 2020-10-12:
Hello all,

for filament change (6in1 diy filament changer with additional extruder to move the filament to direct extruder on print head) i want to use the manual_stepper function in klipper. I want to move the filament with one stepper to the direct extruder and then after the filament had passed a certain distance i need to turn the extruder on the print head in parallel.
But the problem is that i can only move one of them.

Is it possible in klipper to move both in parallel (but only for loading, so the additional steppe synchronized function is not an option).

Thank you for your help!
Matthias
[klippy.log](https://github.com/KevinOConnor/klipper/files/5367026/klippy.log)

theophile, 2020-10-13:
This is exactly what the 'extruder_stepper' config section is for: https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L798 

In fact, I'm using it for the exact scenario you're describing. 6-in-1-out with 2 steppers in each filament path, one of which is a direct-drive extruder. It works great!

modmt, 2020-10-13:
Thank you for the answer. Sounds good. How did you realize that? Because when I use the gcode for example 
G1 E60 F1800 MANUAL_STEPPER STEPPER=stepper_fila MOVE=60 SPEED=30 only the stepper on extruder moves.

I see there is another option: SYNC=0  
But when i use the gcode G1 E60 F1800 MANUAL_STEPPER STEPPER=stepper_fila MOVE=60 SPEED=30 SYNC=0 then I got an "Timer too close" error... 

@KevinOConnor: could you please help regarding the SYNC function?

theophile, 2020-10-13:
> Thank you for the answer. Sounds good. How did you realize that? Because when I use the gcode for example
> G1 E60 F1800 MANUAL_STEPPER STEPPER=stepper_fila MOVE=60 SPEED=30 only the stepper on extruder moves.
> 
> **Edit:** Ah, I see that you use the synchronized stepper function. But that means both steppers will turn parallel all the time, right? In my case I only want to use the first stepper to bring the filament to the second and then the first should be stopped.
> Is it possible to deactivate the synchronisation if it's not needed anymore?

Yes. There's an extended gcode command that will let you change which extruder the extruder_stepper is synced to: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#extruder-stepper-commands 
If you don't want it to move, you can just sync it to an extruder that isn't being used. That said, in your example, if your "first" extruder is going to remain in the filament path, I would recommend against disabling it. I tried that in my case and even the deactivated extruder provided way too much resistance and caused a lot of grinding and underextrusion problems. But having both extruders moving in tandem has been working perfectly for me.

modmt, 2020-10-13:
Thank you. That was a very good idea! 
I added a second extruder (unfortunatelly i have to use 2 free outputs for that, maybe it's possible to define an "dummy" extruder without wasting two pins!?) and switched between the two if I don't need the second with gcode.
In my case I can decouple the second stepper by a servo from the filament path so that there is no resitance from it.
But I will also try to use both in parallel but my fear is that I cannot exactly determine the exact feedrate between the two...

Thank you again. This works great for me so far.

theophile, 2020-10-13:
I was worried about slight differences between the feed rates of the two steppers, but I took the time to calibrate the e-steps for all 7 steppers as precisely as possible. Also, I have pretty long bowden runs between the two steppers in each filament path (about 1 meter or longer), and that helps provide some "wiggle room." After getting the e-steps dialed in, I've never had any problems with the two steppers having different feed rates, even on some 24+ hour prints with no filament changes.

closed 2020-11-04 by modmt