Klipper3d/klipper #3426: Ender 3 SKR Mini E3 | MCU 'mcu' shutdown: ADC out of range

cobramike, 2020-10-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5380317/klippy.log)

When i start my printer its shuting down. Yesterday it was working. And today i was trying to start calibriting Extruder.

MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Klipper state: Not ready

Roycinger, 2020-10-14:
Check / measure if one of your thermistors went bad.
ADC out of range usually indicates that Klipper can't read a value

cobramike, 2020-10-15:
Its working correctly. When i change back to marlin its working like a charm

closed 2020-10-15 by cobramike