Klipper3d/klipper #3427: Invalid speed in 'N975 G1 F0.000*107' Klipper

ChrisBierings, 2020-10-14:
Hi, i was tuning the pressureadvance settings but when i try to do the test print is stops and gives the following error: "Invalid speed in 'N975 G1 F0.000*107" i hope someone knows what the problem is and that they can help me ;)

klipper log file: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5380492/klippy.log)

ChipCE, 2020-10-15:
Just my random guess but doesn't F0.000*107 = 0 ? That mean you request the movement with speed = 0.

mcmatrix, 2020-10-15:
> Just my random guess but doesn't F0.000*107 = 0 ? That mean you request the movement with speed = 0.
> I also cannot find any document about the N975 G-code.

`N975` is just a line number and `*107` is a checksum, Klipper is ignoring these.
https://www.reprap.org/wiki/G-code#Special_fields

ChipCE, 2020-10-15:
@mcmatrix  Thank for point out my mistake! I was searching marlin document for that gcode :) .
But feedrate = 0 is the problem?

ereish64, 2020-10-15:
By setting the feed rate to zero, the printer will not move at all. This is because you are telling the printer it's max travel speed is 0, hence the error.

Personally I consider F1500 to be slow and F3500 to be fast.

ChrisBierings, 2020-10-15:
> By setting the feed rate to zero, the printer will not move at all. This is because you are telling the printer it's max travel speed is 0, hence the error.
> 
> Personally I consider F1500 to be slow and F3500 to be fast.

Where can i change that so i can setup pressure advanced with no problems

Hywelmartin, 2020-10-15:
follow the [guide ](https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md)and slice as usual

closed 2020-11-14 by github-actions[bot]