Klipper3d/klipper #3430: Klipper display config

th3fallen, 2020-10-15:
Per the documentation here https://github.com/th3dstudio/UnifiedFirmware/blob/2.0.x/Firmware/Marlin/src/pins/lpc1769/pins_TH3D_EZBOARD.h#L173

There are no d5,d6,d7 pins so I'm at a loss as to how to map this to a config to make this work any suggestions would be greatly appreciated. this is the LCD in use https://www.th3dstudio.com/product/12864-lcd-display-w-cables-sd-slot-mks-ramps-boards/

[klippy.log](https://github.com/KevinOConnor/klipper/files/5385233/klippy.log)

Hywelmartin, 2020-10-15:
isn't that display the #[ "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays](https://github.com/KevinOConnor/klipper/blob/cf6d302ac3979aec2d55a166fe5446c46c2765ef/config/sample-lcd.cfg#L13)

you need to make(or [buy](https://www.th3dstudio.com/product/ezboard-creality-dual-lcd-adapter-board/)) an adapter that routes all signals needed to EXP1 
Or buy another [display](https://www.amazon.co.uk/creality-Tickas-Printer-Replacement-Creality/dp/B08FHX7MRT)

th3fallen, 2020-10-15:
I have that adapter already installed

Hywelmartin, 2020-10-18:
As THC3 doesn't believe in open source hardware, you have to use a multimeter to see how all things is routed..

Plugs on normal boards
[image]
.
.
.
.
.
Plug on your board
[image]

closed 2020-11-17 by github-actions[bot]