Klipper3d/klipper #3432: Permission denied: '/tmp/printer''

scarecrow2k6, 2020-10-15:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5387316/klippy.log)

I read one or two posts on this issue and one of them did not list the serial correctly (which I did)- https://github.com/KevinOConnor/klipper/issues/1296

The second one was too technical for me to understand- https://github.com/KevinOConnor/klipper/issues/1902

My issue resembles more with the first one. I should mention that I have been using a Rock64 with DietPi->Octoprint instead of a RPi. It worked perfectly well last year, haven't used the printer since March and I think some kind of update might have broken it but I might be wrong. I would really appreciate some feedback. Thanks!

jakep82, 2020-10-16:
The log indicates that it's not seeing your MCU.  

> Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00

What is the output of `ls /dev/serial/by-id/*`

scarecrow2k6, 2020-10-17:
> 
> 
> The log indicates that it's not seeing your MCU.
> 
> > Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00'
> 
> What is the output of `ls /dev/serial/by-id/*`

The output of ls /dev/serial/by-id/*

/dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00

scarecrow2k6, 2020-10-19:
Anyone?

Sineos, 2020-10-19:
I do not know DietPi, so just a wild stab:
- Is serial enabled? Check serial settings in  `dietpi-config` command
- What are the permission of `/dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00`? Show the output of `ls -al /dev/serial/by-id/*`
- Just for testing: Do a `sudo chmod 666 /dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00` (note down the original permissions to revert to, since this is not a long term solution)

scarecrow2k6, 2020-10-21:
> I do not know DietPi, so just a wild stab:
>
>     * Is serial enabled? Check serial settings in  `dietpi-config` command
>
>     * What are the permission of [...]

The output of ls -al /dev/serial/by-id/* is:
/dev/serial/by-id/usb-Klipper_lpc1768_1CB0FF14089DC5AEB375C656C62000F5-if00 -> ../../ttyACM0

I have no idea what that means :(

Sineos, 2020-10-21:
Please post the entire output. Should look like
```
pi@octopi ~ $ ls -al /dev/serial/by-id/*
lrwxrwxrwx 1 root root 13 Oct 18 08:41 /dev/serial/by-id/usb-Klipper_lpc1769_0C70001625813AAF86E06B5CC32000F5-if00 -> ../../ttyACM0
pi@octopi ~ $
```

Under which user do you run klipper? If you do not know, then start klipper and issue the command  `ps axu |grep klipper`. Should look like:

```
pi@octopi ~ $ ps axu |grep klipper
root      6906  0.0  0.1   2144  1308 ?        S    Oct06  17:40 /usr/local/bin/klipper_mcu -r
pi       12845  0.0  0.0   7348   484 pts/2    S+   19:46   0:00 grep --color=auto klipper
pi       32764  3.4  2.6  77020 23816 ?        Sl   Oct18 171:07 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log
pi@octopi ~ $
```

The line
`pi       32764  3.4  2.6  76504 23428 ?        Sl   Oct18 171:12 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer.cfg -l /tmp/klippy.log`
is klipper. The first entry (in my case `pi`) is the user.

When you have the user issue the command `groups your_user_here`. Looks like:
```
pi@octopi ~ $ groups pi
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio
pi@octopi ~ $
```

If you do not see `dialout` in this list, then issue the command `sudo usermod -a -G dialout your_user_here`

Finally log off the shell, log back in and restart klipper.

jchidley, 2020-10-27:
This week I installed Octopi from scratch and ran a `sudo apt update && sudo apt upgrade`.  I created a new user (jack), installed klipper and changed my group ownerships to the same as pi.  

```
jack@octopi2:~/klipper $ groups jack
jack : jack adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio
```

```
jack@octopi2:~/klipper $ ls -l /tmp/printer
lrwxrwxrwx 1 jack jack 10 Oct 26 17:21 /tmp/printer -> /dev/pts/0
jack@octopi2:~/klipper $ ls -l /dev/pts/0
crw-rw---- 1 jack tty 136, 0 Oct 27 10:33 /dev/pts/0
```

I can only connect to `/dev/pts/0` and not `/tmp/printer` (this gives a permissions error) from OctoPrint

closed 2020-11-25 by github-actions[bot]

mberlinger3, 2020-11-25:
I added my $USER (/tmp/printer owner) to the group octoprint and edited my octoprint.service to start as $USER. It defaults to octoprint as the user which cannot access the tmp files created by $USER.