Klipper3d/klipper #3433: Trying to set different colour LEDs

WarriorClub, 2020-10-16:
I am trying to set different colour LED's and tried the codes that include "index".
For example I tried this command
SET_LED LED=my_neopixel BLUE=1.0 Red=1.0 Index=1

But nothing changes.  Shouldn't this change the colour of the first LED to magenta?

mental405, 2020-10-16:
To make the change occur, you have to include the TRANSMIT parameter.

``SET_LED LED=my_neopixel BLUE=1.0 Red=1.0 Index=1 TRANSMIT=1``

WarriorClub, 2020-10-16:
> To make the change occur, you have to include the TRANSMIT parameter.
> 
> `SET_LED LED=my_neopixel BLUE=1.0 Red=1.0 Index=1 TRANSMIT=1`

Thank you.  I also added the transmit parameter but still no changes.

Here is what I added as the code
gcode: SET_LED LED=my_neopixel BLUE=1.0 [INDEX=2] [TRANSMIT=1]

cab404, 2020-10-17:
> > To make the change occur, you have to include the TRANSMIT parameter.
> > `SET_LED LED=my_neopixel BLUE=1.0 Red=1.0 Index=1 TRANSMIT=1`
> 
> Thank you. I also added the transmit parameter but still no changes.
> 
> Here is what I added as the code
> gcode: SET_LED LED=my_neopixel BLUE=1.0 [INDEX=2] [TRANSMIT=1]

Do not include square brackets into your code. They here just to designate that parameter is optional.

WarriorClub, 2020-10-17:
> Do not include square brackets into your code. They here just to designate that parameter is optional.

Thank you!  Both suggestions applied worked.  I will close this issue now.

closed 2020-10-17 by WarriorClub