Klipper3d/klipper #3435: Macro parameter evaluation in conditional

KoeRt888, 2020-10-16:
The following macro does not throw a syntax error for evaluating a macro parameter within a conditional.
```
[gcode_macro START_PRINT]
default_parameter_T_BED: 60
default_parameter_T_EXTRUDER: 200
gcode:
    G90                             ; use absolute coordinates
    M83                             ; extruder relative mode
    G28                             ; cold home all axes
    G1 Z20 F240                     ; raise nozzel off bed
    M190 S{params.T_BED|float}      ; wait for the bed to heat to the desired bed temperature
    {% if params.T_BED >= 100 %}
        HEAT_SOAK
    {% endif %}
    M109 S{params.T_EXTRUDER|float-50}	   ; set the extruder to the first layer temperature - 50 C
    G28                             ; hot home all axes
    G1 Z20 F240                     ; raise nozzel off bed 
    BED_MESH_PROFILE LOAD=HOT       ; load a saved bed mesh profile, in this case the profile for a hot mesh   
    M109 S{params.T_EXTRUDER|float} ; wait for the extruder to heat to the first layer temperature
    PRIME_LINE                      ; call the prime line macro
```

There is no error for using the line ` {% if params.T_BED >= 100 %}`. But if I would use the line `M109 S{params.T_EXTRUDER-50}` there is the hint that the parameter is a string.

Is this due to the mix of macro parameter in jinja conditionals? Shouldn't there be an error to show that the conditional can't execute due to params.T_BED being a string?
I would assume that if I can use the printer pseudo-variable that I could also use the params pseudo-variable.

There is another issue with this macro as can be seen in the log file after the bed has reached the desired temperature. Somehow the conditional messes things up and after heating the bed klipper gives the error 
```
Extrude below minimum temp
See the 'min_extrude_temp' config option for details
```

There is no execution of the M109 command and no extrude command until the PRIME_LINE. The HEAT_SOAK macro just has the toolhead sit in the middle above the bed blasting the fan to heat up the chamber with the radiating heat from the bed. This also happens when I try the line `{% if params.T_BED|float >= 100 %}` or `{% if T_BED >= 100 %}`.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5393336/klippy.log)

Hywelmartin, 2020-10-16:
try

`M109 S{params.T_EXTRUDER - 50}`

KoeRt888, 2020-10-16:
Thats not whats causing the issue.  The expression in the conditional `{% if params.T_BED >= 100 %}` is the line that seems to make trouble.

If I use the line you suggested I get a Klipper error that the arithmetic operation can't be executed since one operand is of the data type string.

Arksine, 2020-10-17:
>There is no error for using the line {% if params.T_BED >= 100 %}. But if I would use the line M109 S{params.T_EXTRUDER-50} there is the hint that the parameter is a string.

Python conditionals will allow comparisons between different types.  Numeric types will always evaluate to be less than strings, so without casting `T_BED` to a float your template will always evaluate that conditional as `True`.

>There is another issue with this macro as can be seen in the log file after the bed has reached the desired temperature. Somehow the conditional messes things up and after heating the bed klipper gives the error

I see that your HEATSOAK macro calls PAUSE.  One thing to keep in mind is that a call to PAUSE does not pause the execution of a macro.  So even if no error was generated, your START_PRINT macro would continue execution after HEATSOAK is called.   This might be the behavior you are expecting, but I thought I would point it out.

The source of your error is likely the retraction done in your PAUSE override, as this executes prior to your M109.  One other thing of note, the internal PAUSE and RESUME commands save and restore gcode state for you.  So while to your calls to save and restore gcode state in your overrides shouldn't hurt anything, they are redundant.

KoeRt888, 2020-10-17:
> The source of your error is likely the retraction done in your PAUSE override, as this executes prior to your M109. 

Thank you for pointing this out. I clearly didn't follow the execution line since I was hung up on the conditional. This is likely the culprit. 
Unfortunately I have a print going right now and can't test the fix of calling something else then PAUSE. But I'm confident that it will fix it.

Thank you and issue closed

closed 2020-10-17 by KoeRt888