Klipper3d/klipper #3436: FR: Bed Mesh Temperature interpolation

cab404, 2020-10-16:
Currently, I have 3 bed leveling profiles for 50, 60 and 70 degrees. Ideally I would like to generate profiles for the missing temperatures automatically, via linear(?) interpolation from existing ones.

Gcode might look something like this:
```
BED_MESH_PROFILE AUTOTEMP=1
```

Looks easy on the surface.

KevinOConnor, 2020-10-19:
I'm sure it's possible to implement something like this.  It would require an interested developer to do the work and submit it.

-Kevin

Coffee0297, 2020-10-20:
Us at the klipper discord have made a macro that did this 
https://klipper.info/macro-examples-1/dutchdudes-automatic-bed-selection

/Tonny

cab404, 2020-10-20:
> I'm sure it's possible to implement something like this. It would require an interested developer to do the work and submit it.
> 
> -Kevin

I would probably try doing this, to get more familiar with codebase)

cab404, 2020-10-20:
> Us at the klipper discord have made a macro that did this
> https://klipper.info/macro-examples-1/dutchdudes-automatic-bed-selection
> 
> /Tonny

I am currently using profiles with temperature names, but that's not what this feature is about. 
It's about generating mesh for 55⁰ from mesh for 50 and 60.

Coffee0297, 2020-10-20:
ah i see. your bed shuld not bend significaly on 5c. unless its hard mounted. but yea then i don't have help for you ;D

/Tonny

closed 2020-11-18 by github-actions[bot]