Klipper3d/klipper #3442: Voron - Failed to verify BLTouch probe is raised; retrying.

paulhew69, 2020-10-18:
Getting this error and not sure why.
2 SKR 1.3's Pi4, Running lastest Klipper - updated yesterday.

Original BL Touch not a clone.
I visually checked it is up and it is, thank goodness!

Running a BLT original on my other printer with Duet boards and never seen this error.

Any suggestions please.
Kind Regards,

Paul.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5397507/klippy.log)

dsteinberg111, 2020-10-21:
on my SKR 1.4 and 1.4turbo, the BL Touch is NOT working like described in the Manual, instead do the 2Pin Connector on the Z-Endstop, where you would normally put the mechanical Endstop. 

The 3 Pin Servo Connector is OK

My Config is then :
[bltouch]
#sensor_pin: P0.10
sensor_pin: P1.27
control_pin: P2.0

My error was alway the last step in the test where mine never triggered on P0.10 no matter what options I tried.

KevinOConnor, 2020-10-22:
Alas, the best information we have on the BL-Touch is at: https://www.klipper3d.org/BLTouch.html 

As indicated above, though, you may have better luck with different pins.

-Kevin

closed 2020-11-09 by paulhew69