Klipper3d/klipper #3443: Can't connect Ender3 V2

yo3gts, 2020-10-18:
Hi, I did everything by the book ... and yet there's something wrong, I can't connect
I have attached the log

[klippy.log](https://github.com/KevinOConnor/klipper/files/5397710/klippy.log)

BlackStump, 2020-10-18:
To me it looks like you have configured the mcu for a BTT Mini E3 V1.2 is that what you have? if you have a standard Ender3 V2 then you should follow the info on that config [here](https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-v2-2020.cfg)

yo3gts, 2020-10-19:
Thank you for the help!
I used this configuration to which I added BL-touch, I also tried display ...
A stupid question ... when I put the klipper.bin firmware on the card and turn on the printer, does it completely replace the marlin firmware and the display is blank? Does the printer only use the clipper firmware?

jerkwagon, 2020-10-31:
@yo3gts did you get this resolved i have the 4.2.7 board as well and im having severe issues. 

as far as i know the screen will go blank unti you plug it in the pie.

closed 2020-11-30 by github-actions[bot]