Klipper3d/klipper #3445: step/dir directly on raspberry pins

etet100, 2020-10-18:
Hi. Is it possible to use klipper with no external cpu? I mean something like this:

https://github.com/markniu/PandaPi

"what's the difference from Klipper

PandaPi: use RPi to control 3D printer directly,except the temperature control which is just to maintain the temperature.

Klipper: uses a RPi to parse G-code,map out curves,set accelerations,and then send the motor command to the MCU via uart."

KevinOConnor, 2020-10-22:
It is technically possible to control a stepper using the Klipper "linux mcu" code.  It's not particularly fast though (see #3387 ).

However, the PandaPi would likely require a notable amount of work because it does have a micro-controller, but it seems that micro-controller may only communicate via i2c.

-Kevin

closed 2020-11-20 by github-actions[bot]

langwadt, 2020-11-20:
maybe with SMI and DMA?

https://iosoft.blog/2020/07/16/raspberry-pi-smi/