Klipper3d/klipper #3446: MCU shutdown: Stepper too far in past.

IAnolastname, 2020-10-18:
So, my printer was working fine (VORON1.0, old iteration of ReFactor that worked). It freaked out on cura 4.7.1 gcode, which I'll attach.

But now since the upgrade to the new version of ReFactor (with new Klipper!), it decided it wants to not print, and throw this error when I attempt to start a print.

I have no idea what's going on.

Posting the .gcode file as well as the .log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5398185/klippy.log)
[benchy1.gcode.txt](https://github.com/KevinOConnor/klipper/files/5398187/benchy1.gcode.txt)

jakep82, 2020-10-19:
You need to use a reasonable max_z_velocity.  If you have leadscrews, it should probably be 25 or less, not 500.  

You need to provide something more descriptive than "it freaked out" for help with anything else.

IAnolastname, 2020-10-21:
That worked fine on previous iterations, hence my confusion.

By "freaked out" I mean steppers spasmed and seemed to not fire properly, then MCU error. I'll cut my Z hop speed down in Cura and see if that fixes it.

jakep82, 2020-10-21:
You risk running into this error again in the future if you don't limit your max_z_velocity.

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis

IAnolastname, 2020-10-21:
Okie doke.

closed 2020-11-20 by github-actions[bot]