Klipper3d/klipper #3447: Malformed command 'PID_CALIBRATE HEATER=ender 3 TARGET=200'

jerkwagon, 2020-10-18:
Brand new setup on  Ender 3 pro, stock. 

i want to calibrate this printer and tune it.   I have made a few test prints, worked fine,  Not much on the web about this error, 

Something about a heater verifier throwing a code. here is my  config as well 

[Here is the config for klipper](https://pastebin.com/wEJ5NxTx)

[klippy.log](https://github.com/KevinOConnor/klipper/files/5398326/klippy.log)
 -->

BlackStump, 2020-10-19:
PID_CALIBRATE HEATER=ender 3 TARGET=200 is Malformed because HEATER=config name of the heater you are trying to tune not the name of your printer!

ie 
extruder or heater_bed
ie 
PID_CALIBRATE HEATER=extruder TARGET=200
or
PID_CALIBRATE HEATER=heater_bed TARGET=60

jerkwagon, 2020-10-19:
Hey thanks,  for some reason i just assumed it was going to save the PID settings in a file or something, this makes so much more sense.   

Maybe we could get a dropdown of the options from the config?  

Thanks again @BlackStump  very helpful!

closed 2020-10-19 by jerkwagon