Klipper3d/klipper #3448: Pressure Advance and Retraction mutually exclusive?

bilalasd, 2020-10-19:
Do I disable retraction in Cura when using pressure advance? The docs are not super clear about that.

The only info I found was this line here https://www.klipper3d.org/Pressure_Advance.html
"Once pressure advance is tuned in Klipper, it may still be useful to configure a small retract value in the slicer (eg, 0.75mm)"

Without pressure advance I have 7mm of retraction on my Tevo Tornado.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5403807/klippy.log)

Hywelmartin, 2020-10-19:
"it may still be useful to configure a small retract value in the slicer"

so do not disable retractions..

bilalasd, 2020-10-19:
Keep retraction enabled but 7mm is too much then right?

bilalasd, 2020-10-20:
also, should the retraction 0.75mm for the testing?

Hywelmartin, 2020-10-20:
7 mm is probably waaaaay long, even without PA... start at 1mm and test if it's enough... Less is more...

dsteinberg111, 2020-10-21:
6mm Retract for PETG is really common. You can retract 2 mm and still see the filament ooze out. PLA is mostly way under 4mm, depends on direct-extrusion, bowden-extrusion and if its a mixing-Hotend.

So If you need 6mm without Pressoure Advance - how much is it then ?

KevinOConnor, 2020-10-22:
This is mentioned briefly in the documentation: https://www.klipper3d.org/Pressure_Advance.html

`The test print is designed to run with a high extruder flow rate, but otherwise “normal” slicer settings. A high flow rate is obtained by using a high printing speed (eg, 100mm/s) and a coarse layer height (typically around 75% of the nozzle diameter). Other slicer settings should be similar to their defaults (eg, perimeters of 2 or 3 lines, normal retraction amount). It can be useful to set the external perimeter speed to be the same speed as the rest of the print, but it is not a requirement.
`

The pressure advance test is largely immune to retraction settings.  If you print normally with 6mm retraction then you can keep it that way for the test.  You can also turn it off for the test.  Once pressure advance is configured I recommend using only a small retraction setting.

FWIW, I find that I need notably higher pressure advance for PETG than I need for PLA or ABS on my printer (eg, 0.138 vs 0.040).  However, I continue to use a small retraction value (eg, 0.75mm) and do not have issues.

-Kevin

gsteffek, 2020-10-24:
> FWIW, I find that I need notably higher pressure advance for PETG than I need for PLA or ABS on my printer (eg, 0.138 vs 0.040). However, I continue to use a small retraction value (eg, 0.75mm) and do not have issues.
> 
> -Kevin

I noticed the same and was thinking it would be nice to have the possibility of setting a PA value via GCode. At least in Prusaslicer it's possible to set a GCode per Filament, so these could be coupled.

Hywelmartin, 2020-10-24:
SET_PRESSURE_ADVANCEADVANCE=<pressure_advance>

richfelker, 2020-10-26:
With a bowden 7mm is very reasonable retraction, especially without pressure advance. I find at least 4-5 mm is preferable even with. Go too low and you'll get stringing/oozing, especially on long travel. It takes nearly 14 mm to get filament out of the heat zone entirely (necessary to prevent oozing on indefinite dwell/travel) on my printer, and anything up to that may be physically justifiable for your needs.

However, due to the necessity of extruder position smoothing with pressure advance enabled, position adjustments due to retraction get incorporated into the smoothing, and this seems to result in erroneous behavior - effectively, retract and unretract happening at the wrong time. I've been experimenting with possible changes and working out the math for exactly what's happening, and I'll open an issue for it when I have enough to make a good report.

With that said, given the current behavior, sticking to the shortest retraction length that works for you and/or slower retraction speeds, and setting the pressure advance smooth time as low as possible, ideally under 10ms if your extruder can handle the rapid accelerations that result, will help a lot.

sonnius, 2020-11-18:
I was looking for methods to find the right retraction values after tuning the PA and came across this:

https://www.thingiverse.com/thing:4532977

would this be a good way of measuring retraction, and if so should it be documented?

closed 2020-12-18 by github-actions[bot]