Klipper3d/klipper #3449: New firmware on Arduino mega and RAMPS 1.4 error

bojfik, 2020-10-19:
I installed Mainsail on my raspberry and after starting the new OS the system informed me that I need to upload the new firmware to the Arduino.  I did it and it turned out that It's not working with RAMPS 1.4
If you do a firmware update on ArduinoMega, in my case with RAMPS 1.4 it won't work. After copying the printer.cfg, it turns out that there are incorrect readings on the housing of the temperature sensors, i.e. on the hotend and the bed. in addition to incorrect temperature readings on the sensors, I noticed that the print cooling fans are constantly on.  in addition to incorrect temperature readings on the sensors, I noticed that the print cooling fans are constantly on.  Which means something is wrong with the new pin map firmware Check it out
I'm going back to Octopi

bojfik, 2020-10-20:
> Hi @bojfik,
>
> It did not look like there was a Klipper log file attached to this ticket. The log file has been engineered to answer common questions the Klipper developers have about the software and [...]

[klippy.log](https://github.com/KevinOConnor/klipper/files/5406199/klippy.log)

bojfik, 2020-10-20:
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/tmp/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.8.0-835-gcf6d302a'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Start printer at Mon Oct 19 19:26:34 2020 (1603131994.2 17.0)
===== Config file =====
[mcu]
serial = /dev/ttyUSB0
pin_map = arduino
restart_method = arduino

[mcu z]
serial = /dev/ttyUSB1
pin_map = arduino
restart_method = arduino

[printer]
kinematics = corexy
max_velocity = 250
max_accel = 2500
max_z_velocity = 25
max_z_accel = 300
square_corner_velocity = 5.0

[idle_timeout]
gcode = 
	TURN_OFF_HEATERS
	M84                            ; disable steppers
timeout = 1800

[stepper_x]
step_pin = ar54
dir_pin = !ar55
enable_pin = !ar38
step_distance = 0.0125
endstop_pin = ^!ar2
position_min = 0
position_endstop = 350
position_max = 350
homing_speed = 100
second_homing_speed = 25
homing_retract_dist = 5
homing_positive_dir = true

[stepper_y]
step_pin = ar60
dir_pin = !ar61
enable_pin = !ar56
step_distance = 0.0125
endstop_pin = ^!ar15
position_min = 0
position_endstop = 350
position_max = 350
homing_speed = 100
second_homing_speed = 25
homing_retract_dist = 5
homing_positive_dir = true

[stepper_z]
step_pin = z:ar54
dir_pin = z:ar55
enable_pin = !z:ar38
step_distance = 0.00250
endstop_pin = ^!z:ar18
position_endstop = 0.85
position_max = 290
position_min = -10
homing_speed = 15.0
second_homing_speed = 1.5
[... 619 lines cut ...]
Stats 55.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000063 bytes_write=5687 bytes_read=7996 bytes_retransmit=0 bytes_invalid=0 send_seq=310 receive_seq=310 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=15891349 z: mcu_awake=0.004 mcu_task_avg=0.000060 mcu_task_stddev=0.000059 bytes_write=1056 bytes_read=5480 bytes_retransmit=0 bytes_invalid=3 send_seq=134 receive_seq=134 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899303 adj=16008090 heater_bed: target=0 temp=-88.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-99.6 pwm=0.000 sysload=0.62 cputime=4.138 memavail=776148
Stats 56.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000063 bytes_write=5751 bytes_read=8130 bytes_retransmit=0 bytes_invalid=0 send_seq=315 receive_seq=315 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.026 ready_bytes=0 stalled_bytes=0 freq=15891347 z: mcu_awake=0.004 mcu_task_avg=0.000060 mcu_task_stddev=0.000059 bytes_write=1062 bytes_read=5538 bytes_retransmit=0 bytes_invalid=3 send_seq=135 receive_seq=135 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899303 adj=16008086 heater_bed: target=0 temp=-38.9 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-63.9 pwm=0.000 sysload=0.57 cputime=4.171 memavail=776148
Stats 57.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000063 bytes_write=5817 bytes_read=8278 bytes_retransmit=0 bytes_invalid=0 send_seq=320 receive_seq=320 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15891346 z: mcu_awake=0.004 mcu_task_avg=0.000057 mcu_task_stddev=0.000055 bytes_write=1068 bytes_read=5623 bytes_retransmit=0 bytes_invalid=3 send_seq=136 receive_seq=136 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899309 adj=16008086 heater_bed: target=0 temp=-88.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-46.9 pwm=0.000 sysload=0.57 cputime=4.206 memavail=776148
Stats 58.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000063 bytes_write=5872 bytes_read=8398 bytes_retransmit=0 bytes_invalid=0 send_seq=325 receive_seq=325 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15891344 z: mcu_awake=0.004 mcu_task_avg=0.000057 mcu_task_stddev=0.000055 bytes_write=1074 bytes_read=5681 bytes_retransmit=0 bytes_invalid=3 send_seq=137 receive_seq=137 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899308 adj=16008125 heater_bed: target=0 temp=-88.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-99.6 pwm=0.000 sysload=0.57 cputime=4.241 memavail=776148
Stats 59.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000076 mcu_task_stddev=0.000063 bytes_write=5936 bytes_read=8532 bytes_retransmit=0 bytes_invalid=0 send_seq=330 receive_seq=330 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=0 freq=15891343 z: mcu_awake=0.004 mcu_task_avg=0.000057 mcu_task_stddev=0.000055 bytes_write=1080 bytes_read=5739 bytes_retransmit=0 bytes_invalid=3 send_seq=138 receive_seq=138 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899307 adj=16008101 heater_bed: target=0 temp=-37.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-54.7 pwm=0.000 sysload=0.57 cputime=4.274 memavail=776148
Stats 60.0: gcodein=0 mcu: mcu_awake=0.007 mcu_task_avg=0.000079 mcu_task_stddev=0.000070 bytes_write=6000 bytes_read=8693 bytes_retransmit=0 bytes_invalid=0 send_seq=335 receive_seq=335 retransmit_seq=0 srtt=0.010 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15891341 z: mcu_awake=0.004 mcu_task_avg=0.000057 mcu_t [... cut ...] 3 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15891324 z: mcu_awake=0.004 mcu_task_avg=0.000057 mcu_task_stddev=0.000057 bytes_write=1206 bytes_read=7065 bytes_retransmit=0 bytes_invalid=3 send_seq=159 receive_seq=159 retransmit_seq=0 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15899307 adj=16008056 heater_bed: target=0 temp=-88.5 pwm=0.000 print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=-63.9 pwm=0.000 sysload=0.41 cputime=4.959 memavail=776216

jakep82, 2020-10-21:
You can't use generic USB serial devices when defining multiple MCUs. They can connect in any order. You need to use by-id or by-path. Read the Klipper installation document to see how to find the proper serial devices names.

bojfik, 2020-10-21:
did you read with understanding what I wrote?  this was configured on the
old Klipper firmware and worked fine unless something changed in the latest
firmware.

jakep82, 2020-10-21:
Nothing has changed with regard to pin definitions for RAMPS.  If it worked previously, it was only because you were lucky and the boards happened to connect in the correct order.  When your main MCU and your Z MCU connect, either one can be assigned USB0, and the other will become USB1.  You need to specify which MCU is which by using specific serial device definitions.  Take a look at the generic Voron config files to see how your MCU serial devices should be defined.

https://github.com/VoronDesign/Voron-2/blob/V2.4/firmware/klipper_configurations/VORON2_stock_250_printer.cfg#L69

bojfik, 2020-10-21:
OK sorry.  my mistake

closed 2020-11-20 by github-actions[bot]