Klipper3d/klipper #3450: filament_switch_sensor not pausing at certain scenarios

ManIkWeet, 2020-10-19:
I have filament_switch_sensor enabled in my config, but it doesn't seem to PAUSE when I start a print without any filament loaded (obviously by mistake, I'm forgetful)
I also noticed it doesn't re-PAUSE when I enter RESUME in the console without any filament present (user error but still).

Is this a config error or is this something that has to be fixed within Klipper?

I currently have a workaround for the start issue, but I doubt it's a great solution:
```
[gcode_macro ENSURE_FILAMENT]
gcode:
    UPDATE_DELAYED_GCODE ID=ENSURE_FILAMENT1 DURATION=0.01
    G4 S1

[delayed_gcode ENSURE_FILAMENT1]
gcode:
    {% if printer['filament_switch_sensor my_sensor'].filament_detected != True %}
        {% if not printer['pause_resume'].is_paused %}
            M118 No filament! Insert some and execute RESUME
            pause
        {% endif %}
        UPDATE_DELAYED_GCODE ID=ENSURE_FILAMENT1 DURATION=0.01
    {% else %}
        M118 Filament present :)
        UPDATE_DELAYED_GCODE ID=ENSURE_FILAMENT1 DURATION=0
    {% endif %}
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/5404488/klippy.log)

Arksine, 2020-10-21:
The way the filament_switch_sensor works is that it pauses when it detects a transition in the switch's state.   There isn't a good way to handle this from within the module itself, as it cannot differentiate between a print and the user issued commands. 

Using a macro is the best solution, however I would do it differently.   Create `CHECK_FILAMENT` macro that you add to the beginning of your "start gcode": 
```
[gcode_macro CHECK_FILAMENT]
gcode:
  {% if printer['filament_switch_sensor my_sensor'].filament_detected != True %}
    action_raise_error("No filament present, aborting print")
  {% endif %}
```
This should raise a gcode error, forcing the print to abort.  
If you want to apply a check to `RESUME`, you could do the following:
```
[gcode_macro RESUME]
rename_existing: BASE_RESUME
gcode:
   {% if printer['filament_switch_sensor my_sensor'].filament_detected != True %}
     action_respond_info("No filament present, aborting resume")
   {% else %}
     BASE_RESUME
   {% endif %}

```
This will override the built-in resume gcode, adding your check.  If no filament is present it will print a message to the terminal, but it will not abort the print, otherwise it will call `BASE_RESUME`, which is what the internal gcode is renamed to.

ManIkWeet, 2020-10-21:
@Arksine I came to the same solution as you provided with CHECK_FILAMENT, works much better indeed.
Although you're missing some {}s around the action_raise_error ;)

That RESUME override is very nice though! I'm going to use that for sure! Didn't know that was possible :D

As for differentiating between print and user issued commands, I'd say that any command involving an extrude should have the filament check, regardless of origin, but maybe that's a bad idea for other reasons.

closed 2020-10-21 by ManIkWeet