Klipper3d/klipper #3451: M109 and M190 S waits are treated as R waits

richfelker, 2020-10-19:
In `klippy/extras/heaters.py`, `check_busy` used absolute value to compare temp with target and also waits for PID loop stabilization. This results in waiting for cooldown as well as heating, which is the function of `M{109,190} Rnnn` commands, not `M{109,190} Snnn` ones. The PID loop wait also makes prints take gratuitously long to start; assuming you'll be doing priming/skirt/etc. there's plenty of time for it to stabilize once printing starts.

Is it desirable to have the `R` behavior as well as `S`? If not, I can submit a trivial PR to make the waits always behave as `S`, but it'll take a little more work to add a distinct `R` form.

richfelker, 2020-10-19:
This issue is filed based on reading of the source, and already contains an analysis of the mechanism of the unexpected behavior. It's not specific to any print/log.

KevinOConnor, 2020-10-19:
This looks like a dup of #1108 .

-Kevin

richfelker, 2020-10-20:
Yep, sorry I missed that.

denunzio11, 2020-10-21:
I left this comment in the referenced closed issue but I'll repeat it here for what it's worth...

I'd like to add my support for this request. As someone who prints commercially, time is a factor. The extra several minutes I have experienced waiting for my printers to start their run is punishing (in the wallet). I know this sounds like an exaggeration but I print objects that have several pieces and some have shorter printing times. Adding minutes to each print adds up...I can only imagine what it would do to a farm environment. Just my 2 cents.

richfelker, 2020-11-12:
Pinging this as not-abandoned. Would a PR that keeps the old behavior as R and changes the behavior for S be open to consideration? If so I'll prepare one soon.

ld3300, 2020-11-12:
I can't test this now as my printer is in the middle of a long print, but this might be doable with just gcode macros.  Something like:

```
[gcode_macro HEATUP]
gcode:
    PAUSE
    M104 S{minimum desired temp}
    UPDATE_DELAYED_GCODE ID=TEMP_CHECK DURATION=5

[delayed_gcode TEMP_CHECK]
gcode:
    {% if printer.extruder.temperature >= (minimum desired temp) %}
        RESUME
    {% else %}
        UPDATE_DELAYED_GCODE ID=TEMP_CHECK DURATION=5
    {% endif %}
```

KevinOConnor, 2020-11-13:
> Would a PR that keeps the old behavior as R and changes the behavior for S be open to consideration?

I commented on this recently at: https://github.com/KevinOConnor/klipper/issues/1108#issuecomment-714018564 .  I think we should move any further discussion to the existing #1108 .

-Kevin

KevinOConnor, 2020-11-17:
Closing this in favor of the existing #1108 .

-Kevin

closed 2020-11-17 by KevinOConnor