Klipper3d/klipper #3453: Bed mesh levelling - nozzle still above bed in centre after G1 Z0

marcosscriven, 2020-10-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5415451/klippy.log)

I've setup a bltouch on an SKR Mini E3 V2.0. The accuracy check shows a s.d of 1 micron:

```
probe accuracy results: maximum 0.160000, minimum 0.155000, range 0.005000, average 0.156750, median 0.157500, standard deviation 0.001601
```

After running a 3x3 mesh level, saving and reloading the config, I then tried running `G1 Z0` at various points on the bed. 

At X0 Y0 the paper test shows it's too close, as the paper is barely moveable. Then at the centre, X150 Y150, the paper test is a little too far away. I move down a further 100 microns, and it's a about right. My understanding is that the probe z_offset might be off if the difference were consistent, but that doesn't seem to be the case.

So I'm not quite clear why - but suspect I'm doing something wrong.

jakep82, 2020-10-21:
It doesn't look like you've calibrated your Z endstop unless it's truly at exactly 0.0.  You might also want to use relative_reference_index.  See the bed mesh document for more details.

https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md

marcosscriven, 2020-10-22:
Thanks @jakep82 

I did look at those docs while levelling the bed. However, I'm still unclear how the Z stop affects or interacts with the mesh?

In the docs for `relative_reference_index` is says:

> the value probed at this coordinate will replace the probe's z_offset

But I thought the z_offset was the difference between the probe tip and nozzle? I don't see how it makes sense to replace that, as it's a fixed thing.

I also don't understand the purpose of measuring the bed with the probe in various places (at which point, with the fixed z_offset you know where the nozzle is), only to then ignore that due to the z height measured by the z-stop?

jakep82, 2020-10-22:
The probe z offset is the distance between the nozzle and the bed when the probe triggers.

marcosscriven, 2020-10-22:
> The probe z offset is the distance between the nozzle and the bed when the probe triggers.

Sorry yes, that's what I mean. When the tip of the probe touches the bed and trigger, you then measure how far down you need to move the nozzle to touch the bed, and that's the offset.

But that should then be fixed right (hence making it a calibration process as in https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md)?

I don't see how the z-stop could then affect that value?

Arksine, 2020-10-22:
The `relative_reference_index` eliminates the need for an exact calibration of the probe's z-offset if the following conditions are met:
1) The printer uses a z-endstop to home the z axis
2) The z-endstop is calibrated against the probe coordinate matching the relative reference index.  So for example, if you specifed a `relative_reference_index` to 4, you would want to calibrate your z endstop position so (160, 140) is at z=0.

The motivation for this is to compensate for probe temperature drift, a common issue with inductive probes.

That said, this may not help your situation.  Some of what you describe appears to be a symptom of [probe location bias](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#location-bias-check), so I would recommend doing the bias check.

marcosscriven, 2020-10-22:
Thanks so much for your help @Arksine - although my printer is Cartesian, I still checked out the probe location bias. But that's prompted more questions:

1. In there it says to go through the `PROBE_CALIBRATE` process in multiple places in the bed. However, so far as I know the accuracy there is not just a function of the probe's consistency in various places, but also how consistently one applies the paper test? I therefore seems easier to me to run `PROBE_ACCURACY` in various places instead - wouldn't that answer the question more easily and accurately?

2. When running `PROBE_ACCURACY` I think I found a bug - the most significant figure seems to increase by one each time:

```
Recv: // probe at 190.000,145.600 is z=2.540000
Recv: // probe accuracy results: maximum 2.545000, minimum 2.537500, range 0.007500, average 2.541250, median 2.541250, standard deviation 0.002016
...
Recv: // probe at 190.000,145.600 is z=3.680000
Recv: // probe accuracy results: maximum 3.680000, minimum 3.675000, range 0.005000, average 3.678000, median 3.677500, standard deviation 0.001500
...
Recv: // probe at 190.000,145.600 is z=4.810000
Recv: // probe accuracy results: maximum 4.812500, minimum 4.807500, range 0.005000, average 4.810250, median 4.810000, standard deviation 0.001750
```

I don't see the probe triggering further and further away, so that looks like a bug.

However, no matter which XY position I run  `PROBE_ACCURACY`, the range reported never goes beyond 7.5um - so I think that rules out probe location bias?

Arksine, 2020-10-23:
Unfortunately the only way to check for location bias is to measure the z-offset at different locations on the bed.   PROBE_CALIBRATE can help with that, but it is still a manual process.  

That said, it certainly appears that you may have found a bug with your PROBE_ACCURACY experiment, and it is possible that is affecting the probing procedure during mesh calibration.  It might be helpful if you can reproduce that, capture the position between calls to PROBE_ACCURACY, then M112 and upload the log, ie: 
```
GET_POSITION
PROBE_ACCURACY
GET_POSITION
PROBE_ACCURACY
GET_POSITION
PROBE_ACCURACY
GET_POSITION
M112
```

There are numerous issues reported regarding the bltouch zoffset and bed_mesh, my guess is this is related.  I don't own a bltouch, but with enough information I may be able to track down what is happening here.  Someone more familiar with the bltouch code may be able to help as well.

marcosscriven, 2020-10-24:
Thanks again @Arksine - reading through some of the other issues it does indeed seem somethings up. 

I doubt there’s location bias as everything seems very square. However I’m away from the printer now for a week so won’t be able to run the test until then. 

Happy to test any bug fixes. 

Thinking about this a bit more, bugs and/or location bias aside, what I’m trying to achieve really is using the z-stop as a backup safety mechanism, and should just use the probe for z-homing as well.

The point being the z-stop can be a little below the bed level, which as it’s on springs can flex a little should the probe fail.

marcosscriven, 2020-10-30:
https://github.com/KevinOConnor/klipper/commit/c2c79ff98d6f5ff0305d69afbbb72a7113ba6251 seems to have fixed this.

Thanks @Arksine / @KevinOConnor

closed 2020-10-30 by marcosscriven