Klipper3d/klipper #3454: Allow gcode macros to call external scripts

abduct, 2020-10-21:
Hello

I think a feature that should be implemented is allowing gcode macros to be able to call external scripts passing data via the command line.

This would open up a vast cove of extra features without much work on the klipper side of things. For example, say I have created a ruby script that uses the a push notification gem that when called will send my phone a notification of printer status obtained from the provided command line arguments. You could then extend various macros to include notification results. This is only one example. You can use this for print statistics, error notifications, ready state notifications, etc.

For example you may be able to extend the END_PRINT macro like so to obtain the first example where a phone would be notified of a finished print:

```
[gcode_macro END_PRINT]
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Move nozzle away from print while retracting
    G91
    G1 X-2 Y-2 E-3 F300
    # Raise nozzle by 10mm
    G1 Z10 F3000
    G90
    # Disable steppers
    M84
   RUN_EXTERNAL_END_PRINT

[gcode_macro RUN_EXTERNAL_END_PRINT]
   RUN_EXTERNAL "ruby ~/scripts/notify.rb 'print finished!'"
```
I am not familiar with python, but if pointed to the right section of the code base, I may be able to figure out a POC for this feature request.

Any suggestions, improvements, or opinions about this feature request would be appreciated.

Thanks/

th33xitus, 2020-10-21:
So you probably mean something like this: https://github.com/KevinOConnor/klipper/pull/2173 ?

ProfEngr, 2020-10-24:
Octorant and a couple of other programs already handle the specific example given. There may be other uses for the expansion though.

abduct, 2020-10-26:
This issue can be closed as it is a duplicate of #2173. I left my remarks and hopefully it can be merged eventually.

closed 2020-11-24 by github-actions[bot]