Klipper3d/klipper #3456: BLTouch v3.1 not reporting probe status corectly after Klipper 0.9 release

MarceloFS13, 2020-10-21:
Hello!

Just updated my SKR1.4 Turbo with the new klipper release and my Bltouch v3.1 (genuine) stopped working correctly.
I have done all the tests suggested on the bltouch docs but it fails when i set BLTOUCH_DEBUG COMMAND=touch_mode and push the pin up send a QUERY_PROBE and the sensors still shows as open. I have checked all the wiring and it seems correct.

MarceloFS13, 2020-10-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5419285/klippy.log)

MarceloFS13, 2020-10-22:
I tried using the probe pins instead of the z-endstop with no success and tried a bltouch clone too and no success

MarceloFS13, 2020-10-22:
Tried a second board and a second set o cables. Tried putting a ^ or ! on the pins but no success. I think its the firmware

mbruszewski, 2020-10-23:
Also having this issues. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5431941/klippy.log)

`# This file contains common pin mappings for the BIGTREETECH SKR V1.3
# board. To use this config, the firmware should be compiled for the
# LPC1768.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: -20
position_min: -20
position_max: 330
homing_speed: 75
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: P1.10
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 250
driver_SGTHRS: 80
diag_pin: P1.29

[stepper_y]
step_pin: P0.19
dir_pin: P0.20
enable_pin: !P2.8
step_distance: .0125
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_endstop: -21
position_min: -21
position_max: 330
homing_speed: 75
homing_retract_dist: 0

[tmc2209 stepper_y]
uart_pin: P1.9
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 250
driver_SGTHRS: 80
diag_pin: P1.28

[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_min: -5
position_max: 400

[tmc2209 stepper_z]
uart_pin: P1.8
microsteps: 16
run_current: 0.650
hold_current: 0.450
stealthchop_threshold: 30

[stepper_z1]
step_pin: P1.15
dir_pin: !P1.14
enable_pin: !P1.16
step_distance: .0025

[tmc2209 stepper_z1]
uart_pin: P1.1
microsteps: 16
run_current: 0.650
hold_current: 0.450
stealthchop_threshold: 30

[heater_bed]
heater_pin: P2.5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.25
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[extruder]
step_pin: P2.13
dir_pin: !P0.11
enable_pin: !P2.12
step_distance: .010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: P2.7
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.24
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 260
pressure_advance: 0.55

[tmc2209 extruder]
uart_pin: P1.4
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 5

[fan]
pin: P2.3

[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_0670000E871C4AAFAB3B7C5DC02000F5-if00
restart_method: command

[printer]
kinematics: corexy
max_velocity: 300
max_accel: 1000
max_z_velocity: 25
max_z_accel: 30
square_corner_velocity: 5.0

###############################################
##  Probing/Mesh
###############################################

[bltouch]
control_pin: P2.0
sensor_pin: P1.27
#pin_move_time: 0.250
pin_up_reports_not_triggered: False
#pin_up_touch_mode_reports_triggered: False
x_offset: -30
y_offset: -10
z_offset: 3.6

[safe_z_home]
home_xy_position: 165,165 # Change coordinates to the center of your print bed
speed: 50
z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[force_move]
enable_force_move: True
 
[bed_mesh]
speed: 120
mesh_min: 5, 5
mesh_max: 325,325
probe_count: 3,3

########################################
# Optional LCD configuration
########################################

# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
[display]
lcd_type: st7920
cs_pin: P1.19
sclk_pin: P1.20
sid_pin: P1.18
encoder_pins: ^P3.26, ^P3.25
click_pin: ^!P0.28
[output_pin beeper]
pin: P1.30

# 128x64 Full Graphic Creality CR10 / ENDER 3 stockdisplay
#[display]
#lcd_type: st7920
#cs_pin: P1.19
#sclk_pin: P1.21
#sid_pin: P1.23
#encoder_pins: ^P1.18, ^P1.20
#click_pin: !P0.28
#
#[output_pin beeper]
#pin: P1.30

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# points =
#*# 	0.155000, 0.305000, 0.387500, 0.552500, 0.545000
#*# 	0.220000, 0.335000, 0.395000, 0.440000, 0.437500
#*# 	0.297500, 0.347500, 0.402500, 0.397500, 0.375000
#*# 	0.307500, 0.360000, 0.392500, 0.362500, 0.275000
#*# 	0.452500, 0.387500, 0.332500, 0.290000, 0.172500
#*# x_count = 5
#*# y_count = 5
#*# min_x = 5.0
#*# max_x = 185.0
#*# min_y = 5.0
#*# max_y = 200.0
#*# x_offset = 27.0
#*# y_offset = 0.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
`

mbruszewski, 2020-10-25:
UPDATE:
I tried various of configs, talked with people on facebook and it went out that bltouch gives wrong statuses, but in the end it works.

Commands I run:
```
Send: BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Recv: ok
Send: QUERY_PROBE
Recv: // probe: open
Recv: ok
[...]
Send: BLTOUCH_DEBUG COMMAND=pin_up
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Recv: ok
Send: QUERY_PROBE
Recv: // probe: open
Recv: ok
```

Bltouch config:
```
[bltouch]
sensor_pin: ^P0.10  # Pull-up (^ symbol) needed in open drain mode
control_pin: P2.0
#pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: False
x_offset: -48
y_offset: -15
z_offset: 3.6
pin_move_time: 0.8
```

Anyway as you can see, the output of QUERY_PROBE is wrong. But anyway it homes successfully.

GrumpaV2, 2020-11-01:
I too had the same problem but moving it to Z- fixed it. All I did was increase pin move time to 0.8. It even reported triggered when running the "basic checks" in klipper documentation. It would not report triggered when "query_probe" connected to "Probe" pin 0.10. I hope this info helps.

closed 2020-11-30 by github-actions[bot]