Klipper3d/klipper #3457: Large Bed Temp Oscillation after PID calibration

denunzio11, 2020-10-22:
I have performed several PID calibrations and it consistently overshoots the bed temperature setting and then continues to fluctuate from, for ex., 68 to 73 degrees (when set to 70).

On the last PID tune I did a WRITE_FILE=1 but I couldn't find it in the tmp dir.

I would suspect hardware issues but I just moved this printer from Marlin where it was rock solid for months.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5419712/klippy.log)

jakep82, 2020-10-22:
In most of your heating attempts the bed was set to watermark.  It's simply shifting between 0 and 1.  It looks like the config was changed to PID, but that was at the end of log, and no further heating attempts were made.

denunzio11, 2020-10-22:
Yep.  I changed it to watermark as an experiment to see if it would perform any better.  I only uploaded a partial log file as it wouldn't allow me to upload a file larger than 10 mb.   I can certainly recreate it a few more times if it would help.

denunzio11, 2020-10-22:
OK in the course of trying to recreate this, I noticed that there was a discrepancy between the printer.cfg that Octoklipper was showing me and the actual printer.cfg on the host.  I have had this instance of Octoprint running for days and at some point it must have...no sure of the word (I'm not a programmer :))...lost sync.  After refreshing the browser, Octoklipper displayed what was actually in the printer.cfg file.  I can only assume that I had one bad PID tune session some time ago and I hadn't been overwriting it when doing my SAVE_CONFIGs.

Ran another PID_CALIBRATE and she is holding steady.  Lesson learned

closed 2020-10-22 by denunzio11