Klipper3d/klipper #3458: X motor does not work good

theodore0310, 2020-10-22:
Hi, after about 30h of printing the x motor started to moves jerking, shake everything and doing a lot of noise. It works as a lot of step which slow down after each one. This is my feeling, please help me. If the log file is not right, let me know. thanks

theodore0310, 2020-10-22:
[serial.log](https://github.com/KevinOConnor/klipper/files/5421436/serial.log)

Roycinger, 2020-10-22:
We would need a klippy.log at least. You can find it in the /tmp folder of your Pi.
See here: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

theodore0310, 2020-10-23:
[Uploading klippy.log…]()

theodore0310, 2020-10-23:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/5428071/klippy.log.zip)
is the same file but compressed

theodore0310, 2020-10-23:
I have an Artillery sidewinder X1 with this profile: https://github.com/Clank50AE/Clanks-Klipper-Configs/blob/master/SideWinder%20X1%20Stock.cfg

theodore0310, 2020-10-28:
the issue was caused by a faulty cable

closed 2020-10-28 by theodore0310