Klipper3d/klipper #3459: Internal error on command:"G1"

Block137, 2020-10-22:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5421858/klippy.zip)

```
...
Loaded MCU 'board2' 82 commands (v0.8.0-811-g08eec0da-20200925_064235-octopi...
...
Stats 176872.7: gcodein=132245 mcu: mcu_awake=0.001 mcu_task_avg=0.000008 mcu_task_stddev=0.000009 bytes_write=200396090 bytes_read=22883948 bytes_retransmit=9 bytes_invalid=0 send_seq=3374093 receive_seq=3374093 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=54 stalled_bytes=0 freq=167998561 board2: mcu_awake=0.000 mcu_task_avg=0.000002 mcu_task_stddev=0.000003 bytes_write=6843263 bytes_read=2363122 bytes_retransmit=9 bytes_invalid=0 send_seq=199100 receive_seq=199100 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=167998726 adj=168000172 sd_pos=12309128 print_time=99287.900 buffer_time=2.018 print_stall=1 extruder: target=200 temp=200.1 pwm=0.441 sysload=0.04 cputime=998.811 memavail=3629200
stepcompress o=4 i=0 c=2 a=0: Invalid sequence
stepcompress o=4 i=0 c=2 a=0: Invalid sequence
stepcompress o=4 i=0 c=2 a=0: Invalid sequence
stepcompress o=4 i=0 c=2 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 177, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 135, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 175, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 419, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 335, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 288, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 171, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Transition to shutdown state: Internal error on command:"G1"
Dumping gcode input 50 blocks
Read 176693.555014: 'M105\n'
...
```

This appears to be duplicate of https://github.com/KevinOConnor/klipper/issues/2793 which is closed (since May) with a solution of updating to newest version however, this happened with Klipper installed last month which I assume has be fixed?

KevinOConnor, 2020-10-22:
Can you provide the gcode file that was being printed at the time of the error?  Can you confirm you set pressure_advance=0.005 and pressure_advance_smooth_time=0.020 sometime prior to the failure?

-Kevin

Block137, 2020-10-22:
> Can you confirm you set pressure_advance=0.005 and pressure_advance_smooth_time=0.020 sometime prior to the failure?

Yes, I did that during 1st layer. Also reduced accel and set feed rate to 60% some time after that.
Shutdown occurred about a day later at 291st layer.

I'll grab the G-code from the Pi tomorrow since I overwrite the original on my PC.

Block137, 2020-10-23:
[Gcode.zip](https://github.com/KevinOConnor/klipper/files/5427438/Gcode.zip)
Shutdown occured at Z=116.4 while printing outer wall.

KevinOConnor, 2020-10-23:
FYI, I am able to reproduce this issue locally.  Early indications are that it is related to pressure advance.  (The print reliably results in an error when PA is enabled.)

-Kevin

KevinOConnor, 2020-10-24:
I think this should now be fixed (commit 1b6b7fc5).

It looks like the problem was related to how pressure advance was calculated.  It was using the absolute E position in its internal calculations.  However, the absolute E position can grow arbitrarily large, and it can saturate the precsion of the hardware floating point registers.  The fix is to use a relative E position for the internal pressure advance calculations.

Unfortunately, this is a pretty bad bug - not only could it cause problems for extremely large prints (as shown here), but it could also occur if many separate prints are made that in total use a large amount of filament.  I'd encourage everyone to upgrade to the code with the fix.

-Kevin

vladbabii, 2020-10-24:
Will this be released as 9.1 ?

KevinOConnor, 2020-10-24:
Yes, I will likely create v0.9.1 because of this error.  However, I'm going to give a week or so to see if any other issues arise before tagging.

-Kevin

closed 2020-11-09 by KevinOConnor