Klipper3d/klipper #3460: Bed_mesh_calibrate/bltouch ignoring z offset - nozzle hits bed on start

jameshgolen, 2020-10-22:
I'm having trouble with my nozzle colliding with the bed when the print starts. I'm trying to use bed_mesh_level and have a bltouch. I've gone through and tried a bunch of different settings but the bltouch z_offset seems to be completely ignored. I have gone so far as to set it to a negative value to see if it actually changes anything. In reality, the offset is about 1.7-1.8mm. Even setting it to -3mm (which should definitely be printing in air) it's still colliding with the bed and doesn't seem any different than if I had it at 1.8mm. I think the problem must either be with my start code, the order of which I'm doing things, or something interal. Any help is appreciated.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5426211/klippy.log)

BlackStump, 2020-10-23:
The z_offset is the distance between the nozzle and bed when the probe triggers. 
>Even setting it to -3mm (which should definitely be printing in air)

 -3mm probe offset is saying that the trigger point is 3mm Below the surface of the bed exactly opposite to your statement.
[you need to follow this doc to get the correct probe offset](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-z-offset)

CHRSalakev, 2020-10-23:
I had the same issue =)

In marlin you use Z_offset = -1.3 
In Klipper its reversed, z_offset: 1.3

ksalaba, 2020-10-23:
I had the same issue =)

In marlin you use Z_offset = -1.3
In Klipper its reversed, z_offset: 1.3

jameshgolen, 2020-10-23:
I disagree. As proof, I have set the offset to +5mm. I used a 1cm block to make sure it wouldn't go through my bed into oblivion. If I wouldn't have e-stopped, there's a good chance it still would have went through. First shot I had to pull the plug since it was about to touch. It may not be being "ignored" but something is not being calculated correctly either due to a setting in my config, start code, or something else.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5430648/klippy.log)

Hywelmartin, 2020-10-23:
[stepper_z]
...
...
position_min = -5  set to 0

[printer]
...
...
minimum_z_position: -5  Add this

jameshgolen, 2020-10-23:
minimum_z_position is not a valid setting? I don't even see that option in the example config files. Would it possibly be something like the following?

[stepper_z]
position_min = -5
endstop_position = 0

Arksine, 2020-10-23:
After looking through @jameshgolen's second log, I think that I may have discovered an issue.  I am not extremely familiar with the bltouch code, so perhaps @KevinOConnor or one of the bltouch contributors can verify my findings.  The serial dump starts at line 1196, and at line 1238 an `endstop_home` command is queued.  This appears to occur after the z axis has been commanded to lift.   Below are the last three `endstop_state` responses:
```
Receive: 16 591.943103 591.939747 9: seq: 1f, endstop_state oid=15 homing=0 pin_value=1
Receive: 17 591.945866 591.945262 9: seq: 11, stepper_position oid=7 pos=4946
Receive: 18 591.947247 591.946938 9: seq: 12, stepper_position oid=8 pos=4946
...
Receive: 32 593.748629 593.699297 9: seq: 14, endstop_state oid=15 homing=0 pin_value=1
Receive: 33 593.752813 593.751605 9: seq: 17, stepper_position oid=7 pos=4948
Receive: 34 593.754116 593.753708 9: seq: 18, stepper_position oid=8 pos=4948
...
Receive: 43 594.687387 594.428763 9: seq: 1b, endstop_state oid=15 homing=0 pin_value=0
Receive: 44 594.690100 594.689037 9: seq: 1d, stepper_position oid=7 pos=5804
Receive: 45 594.691483 594.691051 9: seq: 1e, stepper_position oid=8 pos=5804
```
The values received in the last pair of `stepper_position` responses deviate quite a bit from those received previously.

A call to the `BLTouchEndstopWrapper.verify_state()` method seems to be the source of the final `endstop_home` command.   My assumption is that we do not want `MCU_Stepper.note_homing_end()` called after `endstop_state` is received here, as `verify_state()` doesn't actually home a stepper.   If I am correct, a fix could look something like this:
```diff
diff --git a/klippy/extras/bltouch.py b/klippy/extras/bltouch.py
index a73e9ab49..978abee01 100644
--- a/klippy/extras/bltouch.py
+++ b/klippy/extras/bltouch.py
@@ -114,7 +114,8 @@ class BLTouchEndstopWrapper:
         self.mcu_endstop.home_start(self.action_end_time, ENDSTOP_SAMPLE_TIME,
                                     ENDSTOP_SAMPLE_COUNT, ENDSTOP_REST_TIME,
                                     triggered=triggered)
-        return self.mcu_endstop.home_wait(self.action_end_time + 0.100)
+        return self.mcu_endstop.home_wait(self.action_end_time + 0.100,
+                                          notify_steppers=False)
     def raise_probe(self):
         self.sync_mcu_print_time()
         if not self.pin_up_not_triggered:
diff --git a/klippy/mcu.py b/klippy/mcu.py
index df49c9275..26da71c24 100644
--- a/klippy/mcu.py
+++ b/klippy/mcu.py
@@ -99,13 +99,14 @@ class MCU_endstop:
             if est_print_time >= self._next_query_print_time:
                 self._next_query_print_time = est_print_time + self.RETRY_QUERY
                 self._requery_cmd.send([self._oid])
-    def home_wait(self, home_end_time):
+    def home_wait(self, home_end_time, notify_steppers=True):
         self._home_end_time = home_end_time
         did_trigger = self._home_completion.wait()
         self._mcu.register_response(None, "endstop_state", self._oid)
         self._home_cmd.send([self._oid, 0, 0, 0, 0, 0])
-        for s in self._steppers:
-            s.note_homing_end(did_trigger=did_trigger)
+        if notify_steppers:
+            for s in self._steppers:
+                s.note_homing_end(did_trigger=did_trigger)
         if not self._trigger_completion.test():
             self._trigger_completion.complete(False)
         return did_trigger
```

KevinOConnor, 2020-10-24:
Thanks @Arksine .  I think you're correct and that there is a race between the lifting of the toolhead and the verification of the bltouch pin when `stow_on_each_sample=False`.

I think the fix would just be:
```diff
--- a/klippy/extras/bltouch.py
+++ b/klippy/extras/bltouch.py
@@ -173,6 +173,7 @@ class BLTouchEndstopWrapper:
     def multi_probe_end(self):
         if self.stow_on_each_sample:
             return
+        self.sync_print_time()
         self.raise_probe()
         self.sync_print_time()
         self.multi = 'OFF'
```

-Kevin

Hywelmartin, 2020-10-24:
@jameshgolen  the min position under [printer] is a delta only thing.. sorry for that.. but state the endstop position  of else it will be the default 0.5.. and might confuse some people..
as kevin pointed out and fixed(hopefully) there was some fault in the code..

Arksine, 2020-10-24:
Thanks Kevin, that makes sense.

KevinOConnor, 2020-10-24:
FYI, I committed the fix for `stow_on_each_sample=False` (commit c2c79ff9).

-Kevin

jameshgolen, 2020-10-24:
I'll pull and give it a try, thanks guys

jameshgolen, 2020-10-24:
Looks like this fixed it! Thanks!

closed 2020-10-24 by jameshgolen