Klipper3d/klipper #3461: SKR Mini E3 1.2 Latest firmware 'bricks' board... ?

phuzzyday, 2020-10-23:
So I know everyone is going to ask me all the basic questions, but I have been using klipper for a couple years or so, with great success, so I THINK I did everything right!

PROBLEM: 
- Used 'make config', set all options as instructed in the config template for the SKR Mini E3 1.2.  )And checked many times now.  And re compiled.  and copied using the PI itself to an sd card, same result as PC..)  (Checked Bootloader 'offset'?, and extra pin action mentioned in template also.)

- Right after flashing the klipper.bin, (rename, sd card, fat32), Controller no longer shows up on the Raspberry pi.  Or a PC if you plug it in.  Nothing.  Nada.  I have to flash the STOCK firmware to it, which contains all the other partitions I think, to get it showing back up on USB.  

STORY:
I decided to make a completely new octoprint installation, for..  reasons..  Anyway, I used another SD card, still have both.  

Everything went smooth until I tried to SEE the controller with klipper, and nothing.  I thought perhaps the ID had changed or something, so I looked at /dev/serial/by-id, and it didn't exist.  

I can recover it with the stock firmware, and flash the previous version I was using, recompiling the SAME way, and get back in action. Version from 2 days after klipper 0.9.0 came out.

I have been at it for MANY HOURS.  Reflashing over and over, recovering over and over, different SD cards, and as I mentioned, using just using the Raspbian GUI to move and rename the klipper.bin file.  (I wonder.  Maybe there is a new stock firmware for the SKR with newer versions of the rest of the partitions that may help...  I'll try that next.

For now, I am running the WORKING one, which this log is from.  I don't think the log is relevant, there may be another one from the other installation someone needs though.. Complilng log?

Block137, 2020-10-23:
With my small experience, after firmware has been flashed, remove the SD card and use without it.

phuzzyday, 2020-10-23:
> With my small experience, after firmware has been flashed, remove the SD card and use without it.

This guy renames the firmware file after it's done, and it doesn't see it after that on reboot.  It works either way, and I have done both with this issue.  No diff.

kookaburra88, 2020-10-23:
Same problem here since firmware update to klipper 0.9

phuzzyday, 2020-10-24:
The GPIO pin thing is still there when you enable advanced options.  Glad it's not just me!!  I DID try to find other threads...
I'm back in action anyway...
I managed to yank the klipper files from the old installation and into the new.  Once I learned about making files executable everything went fine.  (I pulled the files out of a backup archive of the card I made using the Ext2fsd driver in Windows and FTP'd them back into the Pi.  That probably screwed up the permissions.  Maybe.)

kookaburra88, 2020-10-24:
@phuzzyday Thanks, I did not realized that I have to use keyboard y/n to enter the advanced section. I got my two SKR mini V1.2 running now on Klipper 0.9 with configuration described as in https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v1.2.cfg

1N4148, 2020-10-24:
Really checked the 28kib bootloader? Maybe you can post your .config here?

kookaburra88, 2020-10-24:
Sure, here are my config and firmware.bin:
[image]

[firmware.zip](https://github.com/KevinOConnor/klipper/files/5432729/firmware.zip)

1N4148, 2020-10-24:
Strange. I've made firmwares for my SKR E3 Mini V1.2 and V2.0 a few days ago and everything is working fine. The only changes in source are the updated bootloader offsets.

[firmware_v12.zip](https://github.com/KevinOConnor/klipper/files/5432749/firmware_v12.zip)

1N4148, 2020-10-24:
I've built it again with actual branch/commit, working fine on my Ender 3 (SKR Mini E3 v1.2) and Ender 5 (SKR Mini E3 V2.0). 

[BTT_SKR_MINI_E3_V12_V20_KLIPPER_v0.9.0-0-g2bcf06a2.zip](https://github.com/KevinOConnor/klipper/files/5432795/BTT_SKR_MINI_E3_V12_V20_KLIPPER_v0.9.0-0-g2bcf06a2.zip)

phuzzyday, 2020-10-26:
Those settings match mine.  Bootloader and GPIO Pin also.  Lots of people talking about v2.0 boards working..  This is 1.2, there are differences surely!

kookaburra88, 2020-10-26:
@phuzzyday at least two people here (@1N4148  & myself) have reported working SKR mini V1.2. 
Have you tried the attached binaries?

1N4148, 2020-10-26:
The only important difference when building the firmware is the custom pin to set at startup.
v1.2 needs !PC13, v2.0 needs !PA14. But this is important as these pins are controlling the USB data line pullup. Without the host isn't detecting the board.

phuzzyday, 2020-10-31:
Will Test those binaries to see what happens!

closed 2020-10-31 by phuzzyday