Klipper3d/klipper #3463: Reposting Temperature Reactive LED

WarriorClub, 2020-10-24:
I am trying now to include the temperature reactive LED's to printer config. I have added a separate LED config file [tempreactLED.cfg]" to the config and included the codes from Mental on "tempreactLED.cfg".
It says the tempreactLED.cfg is not valid. Not sure why it is not working.  

klippy.log
I hope this is the right way to attach the log. Could not find clearer instructions on how to.

WarriorClub, 2020-10-24:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5433085/klippy.log)

theopensourcerer, 2020-10-24:
```
MissingSectionHeaderError: File contains no section headers.
file: /home/pi/tempreactLED.cfg, line: 2
'M105.1\n'
File contains no section headers.
file: /home/pi/tempreactLED.cfg, line: 2
```

WarriorClub, 2020-10-24:
S

> ```
> MissingSectionHeaderError: File contains no section headers.
> file: /home/pi/tempreactLED.cfg, line: 2
> 'M105.1\n'
> File contains no section headers.
> file: /home/pi/tempreactLED.cfg, line: 2
> ```

Sorry.  I am still new to all this.  What is a section header?

theopensourcerer, 2020-10-24:
It's the bit in the square brackets, e.g:

[printer] or [neopixel my_neopixel].

Try posting the contents of your tempreactLED.cfg file. because klipper can't read it none of your config is shown in the log (which it normally would be) ;-)

WarriorClub, 2020-10-24:
Sorry, for some reason it would not allow for cfg file so I changed it with log as extension.  Here is my tempreactLED
[tempreactLED.log](https://github.com/KevinOConnor/klipper/files/5433118/tempreactLED.log)

theopensourcerer, 2020-10-24:
I'm not familiar with what you are trying to achieve but the first section header doesn't look correct to me. Just "[macro]" is not valid I believe... See the examples for gcode macros:

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L396

WarriorClub, 2020-10-24:
I am just trying to get the LEDs to change colour depending on the temperature of the nozzle.  I got this code from Mentals https://klipper.info/macro-examples-1/mentals-temperature-reactive-leds.  I will have to take a closer look at the appropriate section headers for this but for now here the full log.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5433154/klippy.log)

theopensourcerer, 2020-10-24:
Just delete the first three lines of your file. The author was explaining what the output will be, not that you were supposed to copy that part. Your file should look like this I believe:

```
[neopixel temp_leds]
pin:                                P1.18 
chain_count:                        1

[gcode_macro M105]
rename_existing:            M105.1
gcode:  

   M105.1

   #if the extruder is off
   {% if printer.extruder.target == 0 %}

      #Set the LED to red if the extruder is off but is still hot, otherwise 
      # set the color to green
      {% if printer.extruder.temperature > 60.0 %}
         SET_LED LED=temp_leds RED=1 GREEN=0 BLUE=0 INDEX=0  TRANSMIT=1
      {% else %}
         SET_LED LED=temp_leds RED=0 GREEN=1 BLUE=0 INDEX=0  TRANSMIT=1
      {% endif %}     

   {% else %}

         #if the extruder temp is at target temperature 
         {% if printer.extruder.temperature >= printer.extruder.target - 4.0 %}
            SET_LED LED=temp_leds RED=1 GREEN=0 BLUE=0 INDEX=0  TRANSMIT=1

         #if the extruder is still heating
         {% else %}
            {% set scaler = printer.extruder.temperature|float / printer.extruder.target|float %}
            SET_LED LED=temp_leds RED={ scaler|float * 1 } GREEN=0 BLUE=0 INDEX=0  TRANSMIT=1
         {% endif %}  

   {% endif %}
```

WarriorClub, 2020-10-24:
Thank you. I omitted those lines but still no luck.
Not sure what the pin: 1.18 is.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5433816/klippy.log)

closed 2020-11-22 by github-actions[bot]