Klipper3d/klipper #3466: DELTA_CALIBRATE question

toskium, 2020-10-24:
I am running a relatively tall delta printer (1m towers) due to thermal expansion I regularly have to rerun DELTA_CALIBRATE.

I do have a rather precise Z-probe (Duet Smart Effector) which is why I am relying on the automatic process. 

From my understanding DELTA_CALIBRATE invalidates the values of the enhanced delta calibration and needs to be run again as well.

Is there a less time consuming way of handling this issue?

Cheers, Markus

Hywelmartin, 2020-10-24:
the expansion should be equal.. so try to just set a new bed height..

```
 
[gcode_macro set_delta_height]
gcode:
	M117 SET_DELTA_HEIGHT	
	G90	
	G0 F8000 X0 Y0 Z10 
	PROBE
	G92 Z-0.12  #probe offset
	G0 F5000 Z10
```

toskium, 2020-10-25:
Thanks for your suggestion. I am currently doing some tests and will include your macro code.

toskium, 2020-10-25:
Okay, I rebuilt my start gcode macro to account for the expansion and it seems to be working fine now. Thanks for the help!

closed 2020-10-25 by toskium